mathoudebine/turing-smart-screen-python #6: Screen orientation

alhemicar86, 2022-01-15:
Hi, first of all great work with a script, works really well and helps understand the screen operation.

I noticed a screen orientation issue, but not sure if it is screen problem or script problem: If I kill the running script from python and then run it again, the screen will render everything in protrait orientation instead of vertical on second and each next run, unless the screen itself is unplugged and plugged back in before running the script again.

mathoudebine, 2022-01-26:
Hi,
I did not notice this kind of behavior, but as a quick fix you may try to call the `Reset(lcd_comm)` function https://github.com/mathoudebine/turing-smart-screen-python/blob/deb0a60b772f2c5acef377f13b959632ca649f9f/main.py#L42
You can call it instead of the `Clear(lcd_comm)`
It should reset all screen parameters, like if the screen was unplugged then plugged again.
Please keep me in touch

alhemicar86, 2022-01-26:
Hi,
I tried it, after replacing Clear with Reset command it behaves the same way as it did with Clear, however, if I add new line with Reset before the Clear, the COM port disconnects, it clears the screen and the script stops the execution with following error:

> Python 3.10.1 (tags/v3.10.1:2cd268a, Dec  6 2021, 19:10:37) [MSC v.1929 64 bit (AMD64)] on win32
> Type "help", "copyright", "credits" or "license()" for more information.
> 
> === RESTART: C:\Users\iv\Downloads\turing-smart-screen-python-1.1\main.py ===
> Traceback (most recent call last):
>   File "C:\Users\iv\Downloads\turing-smart-screen-python-1.1\main.py", line 188, in <module>
>     Clear(lcd_comm)
>   File "C:\Users\iv\Downloads\turing-smart-screen-python-1.1\main.py", line 47, in Clear
>     SendReg(ser, Command.CLEAR, 0, 0, 0, 0)
>   File "C:\Users\iv\Downloads\turing-smart-screen-python-1.1\main.py", line 39, in SendReg
>     ser.write(bytes(byteBuffer))
>   File "C:\Users\iv\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialwin32.py", line 325, in write
>     raise SerialTimeoutException('Write timeout')
> serial.serialutil.SerialTimeoutException: Write timeout

After that, retuning the Clear line back where it was, and removing the Reset rotates the screen properly on the next run.

alhemicar86, 2022-01-26:
Got it working with this code:

    lcd_comm = serial.Serial(COM_PORT, 115200, timeout=1, rtscts=1)

    Reset(lcd_comm)
    sleep(0.5)
    
    lcd_comm = serial.Serial(COM_PORT, 115200, timeout=1, rtscts=1)

    Clear(lcd_comm)

After invoking reset it needs time to reconnect to the pc, and also something happens in the background, so it needs to define serial port again. My Python skill is limited, so I do not know if this may cause different problems further down the line.
Tried the above code with lcd_comm.close() followed lcd_comm.open(), after the sleep, but it, if it doesn't fail, has the same rotation issue.

mathoudebine, 2022-01-27:
If i remember well, there is a command to set the screen rotation (that is not yet implemented in this project).
What I think about your issue: when you stop your code, the last frame sent to the screen is not complete, and the screen interprets it as a "rotation command".
Don't know if this is the case for your screen only or if others have this issue.

What you implemented is good: this way you are sure that the screen is always reset before running the program.
As you found out, the reset disconnects then reconnects the screen so you have to add a delay after.

Another option would be to reset the screen at the end when you stop the program, just before the serial port is closed:
```
    Reset(lcd_comm)
    lcd_comm.close()
```
So when you start the program again, the screen is already reset.

mathoudebine, 2022-01-27:
Also, just to be sure the last frame is not cut at stop, you have to stop the program with SIGINT (Ctrl + C), SIGTERM or SIGQUIT (Linux only). Avoid sending SIGKILL (kill -9 ...)

alhemicar86, 2022-01-27:
I did notice that if I run from Command Prompt, and terminate it with SIGINT (Ctrl + C) it will keep the proper orientation after script restart. Trouble seems to be with the Python3 IDLE which, I guess, sends SIGKILL to the process by default. Also, just closing the Command Prompt yields the same result. In both cases SIGKILL breaks execution mid loop.

Is there some manual with list of commands for these screens?

mathoudebine, 2022-02-03:
Unfortunately no command lists, all the commands implemented in this project come from the original binary of the vendor, that can be decompiled with a tool like JetBrains dotPeek and then loaded like a VS project (decompiling a software may be illegal in your country, check your local regulations)

I found the rotation command not implemented tin this project, here is the original C# code:
```c#
public void SetRotation(int type, int width, int height)
{
    type += 100;
    byte[] byteBuffer = new byte[16];
    byteBuffer[6] = (byte)type;
    byteBuffer[7] = (byte)(width >> 8);
    byteBuffer[8] = (byte)(width & (int)byte.MaxValue); // byte.MaxValue = 255
    byteBuffer[9] = (byte)(height >> 8);
    byteBuffer[10] = (byte)(height & (int)byte.MaxValue); // byte.MaxValue = 255
    this.SendReg(121, 0, 0, 0, 0, byteBuffer);
    Thread.Sleep(10); // useful?
}
```
Parameters of this function are:
- type: [0-3] (landscape, reverse landscape, portrait, reverse portrait)
- width of the LCD
- height of the LCD

If the screen is set in landscape, you have to swap the width/height value

If you want to implement this command in Python, i would gladly review your PR.
Otherwise, i will implement it when i have some spare time.

alhemicar86, 2022-02-09:
Hey, sorry for late response.
Busy with work last few weeks, but will try it once I get some time.

alhemicar86, 2022-02-21:
I think I have working code:

```
def SetRotation(ser: serial.Serial, rot: int):
    rot += 100
    width = DISPLAY_WIDTH
    height = DISPLAY_HEIGHT
    x = 0
    y = 0
    ex = 0
    ey = 0
    byteBuffer = bytearray(11)
    byteBuffer[0] = (x >> 2)
    byteBuffer[1] = (((x & 3) << 6) + (y >> 4))
    byteBuffer[2] = (((y & 15) << 4) + (ex >> 6))
    byteBuffer[3] = (((ex & 63) << 2) + (ey >> 8))
    byteBuffer[4] = (ey & 255)
    byteBuffer[5] = Command.SET_ROTATION
    byteBuffer[6] = rot
    byteBuffer[7] = (width >> 8)
    byteBuffer[8] = (width & 255)
    byteBuffer[9] = (height >> 8)
    byteBuffer[10] = (height & 255)
    ser.write(bytes(byteBuffer))
    sleep(0.5)
```

Also added 
`    SET_ROTATION = 121`
under the Command class

And after reseting and clearing scree 
```
    # Set screen orientation
    SetRotation(lcd_comm, 1)
```
which inverts it upside down

alhemicar86, 2022-02-21:
Should probably add exchange of width and height in if the rot is greater than 101, though I didn't see any difference, but might be some. Also didn't try making example landscape friendly, for now I will guess it may work if images and elements are given proper landscape coordinates. I can send you the py file I worked on, though I ruined it a bit while trying stuff.

mathoudebine, 2022-03-14:
Hello, you can send the file and i will try to integrate it when i have some spare time!
Thanks for writing the code

alhemicar86, 2022-03-15:
File sent

closed 2022-03-15 by alhemicar86