mathoudebine/turing-smart-screen-python #7: Screen displays images but is corrupted and weird.

opless, 2022-02-02:
Someone else has had a similar issue here:

https://youtu.be/4z5OOW1jL94

I’m also using an M1 Mac.

I’m also reverse engineering the vendor .net binaries to attempt to figure out what is going wrong.

Though I’m struggling a little with the obfuscation. 😂

it’s not something as simple as endian issues!

opless, 2022-02-03:
Looks like it's something weird with the serial comms or the operating system.

Closing because it's not fixable here.

mathoudebine, 2022-02-03:
Sorry to hear that, FYI the display you should expect is this one: https://www.youtube.com/watch?v=Y1m4KK881Aw
I had greater success in decompiling the chinese binary than the english one, it was not obfuscated. You will find the link of it in the README

opless, 2022-02-03:
I've had a look at all three, and yes, the one you mention is MUCH easier to work with.

It's a shame there's no way to get hold of the creators and have a look at their firmware source.

I'm considering knocking something up in golang, to see if it's just the serial port drivers are broken on the M1 or if it's just pyserial that's not working.

alanrenouf, 2022-05-13:
Im also getting this issue on my Mac, dont get it on windows.  the port shows up as "/dev/tty.usbmodemUSB35INCHIPSV21" @opless did you find a workaround ?

opless, 2022-05-13:
It's very much to do with the USB serial implementation, as far as I can tell.

I had some better luck sending data to it via golang, as the serial port initialisation is different. 

I pretty much put the project aside after things getting busy with the day job and having less brain in the evening. It's certainly something weird with the M1 serial usb drivers though!

mathoudebine, 2022-05-16:
Hi, i'm reopening this issue for people with Mac that have the same problem. Feel free to discuss here. What you can also try is changing the COM port settings here: https://github.com/mathoudebine/turing-smart-screen-python/blob/deb0a60b772f2c5acef377f13b959632ca649f9f/main.py#L184
If i remember correctly, the display also works with baudrates slower than 115200

Ebag333, 2022-06-06:
https://github.com/mathoudebine/turing-smart-screen-python/pull/15

I think I've fixed it by serializing the updates to the screen (using a queue, so it'll run through each one cleanly before executing the next).

With the default code base I could get it to step on itself, once I started adding a bunch more things to update it just got worse and worse.  You could help by increasing the delay after an update was done, but slowing down the updates isn't really a solution as it makes everything slow.

With this PR it runs at a very high rate and has no issues with keeping up, not getting serial errors anymore.

gerph, 2022-08-29:
From my own investigation:

In my testing in my own library, I found that the delay between the bitmap data being output and the next command being issued needs to be quite long - about 0.02 seconds. Shorter than this and you get corruption on the display. The current delay in the code on the master branch is about 0.01 seconds, which means that the other slack is being taken up by the code sitting between the `time.sleep(0.01)` and the next `SendReg` writing a command. On faster machines that code will be executed faster, and result in the corruption (I believe).

In my library I changed the behaviour so that instead of explicitly sleeping at the end of the update of a bitmap, we remember when the last output happened and if we try to write a command before the required inter-bitmap-delay has expired, we sleep there instead. This means that we get useful work done between the two operations, and if you did enough work you'd never need to sleep at all before the next command was written.

My library changes that did this were here, if they're useful to you:

https://github.com/gerph/turing-smart-screen-python-mit/commit/8ba5901224ba5e52c37e86e668f73eca4d890b4d

It looks like the changes in PR #15, mentioned above, move the logic for the delay out of the communications library. I cannot tell how this keeps the bitmap transfers and the subsequent command updates apart enough to avoid the corruption.

gerph, 2022-09-01:
I've tried out the feature/system-monitor branch, at f3748418c42, with flagship display, on my macOS system... I tried running the main a few times, and observed corruption on each of them - the first run generally doesn't even display the main image. The main image also appears offset for some reason.

https://user-images.githubusercontent.com/13813876/187958553-226441df-aeb4-486d-be07-a797d872448a.mp4

I'm not sure why it's doing that, but that's what I've seen with it so far - I may try and investigate later.

I suspect there's some race in the lcd_comm library - looking at lcd_comm_reb_b.py line 197, I see that we issue a SendCommand DISPLAY_BITMAP without the bypass_queue, and then go into loading the image and try to send it. Because bypass_queue wasn't set, the DISPLAY_BITMAP may not have been sent yet and the data for to send to the device may have been may get sent first. Additionally, the WriteData command *isn't* protected by the mutex, so it's possible that the queued data gets send DURING the image bitmap data.

gerph, 2022-09-01:
Oh, I'm wrong about the data being sent during the bitmap update - SendLine is a queue operation, so it should be queued after the DISPLAY_BITMAP command. However, I think there's still a possibility that the SendCommand could be issued then something else get inbetween it and the data from the scheduler.

I'm really not at all convinced that this queue is the right way to go - for a start the queue code mixes the application-specific clients with the general queue dispatch, which means that if you want to re-use this code you have to copy out the queue handler code. And in any case, I'm not sure that it helps with any corruption problems at all.

gerph, 2022-09-01:
I've tried replacing the queuing code with a simpler mutex-around-atomic operations with delay between the bitmap and next command. This has been solid on my flagship device. Unfortunately I cannot speak to the model A device... but I've tried making the necessary changes - it might not work at all on those models, and there might be a completely different processing model causing the problems there.

Here's the video of it working (again, tested 4 times to be sure it's not a fluke):

https://user-images.githubusercontent.com/13813876/188019709-de91bbd3-a9c6-45a7-8725-8c15cda36e8c.mp4

PR raised with the way that it worked for me. I've ordered a model A, but it'll be a while until it gets here.

https://github.com/mathoudebine/turing-smart-screen-python/pull/34

gerph, 2022-09-01:
It still might be useful in some cases to have the serial comms on a separate thread, so it's possible the queue could be useful for that. However, that is still a bit complicated, and I didn't attempt it.

Bobrobot1, 2022-09-12:
I have this issue as well, M1 Pro

SaplingLTD, 2022-11-13:
Im facing same problem with the last version in MacOS. Trying to figure out why it happened.
[image]

gerph, 2022-11-13:
Oooh, it's built into a keyboard?! Is that a custom build or something that others can get hold of? (sorry I still don't know why there's corruption, though :-( )

SaplingLTD, 2022-11-14:
So far, I can run the python code normally to connect and give command to the screen.
I also tried to use this software on Windows https://wiki.keebmonkey.com/en/products/kbmscreen. It is recommended by the keyboard marker. And it worked fine.

mathoudebine, 2022-11-14:
Hi @SaplingLTD and welcome to this project

We didn't know the screen had been embedded into a keyboard! Looks really nice
Can you try to use the branch `fix/replace-queued-data-send-with-immediate-plus-deferred-delay` instead of `main` branch? It should solve the issue with MacOS

It is actually in PR #60 because of issues when the program stops: sometimes after a relaunch the screen is in an invalid state and the program crashes. When relaunched again it will then work.

amiltonjr, 2023-01-07:
@mathoudebine I just tested the branch `fix/replace-queued-data-send-with-immediate-plus-deferred-delay` in MacOS Ventura 13.1 with a `Revision A` display and it did not corrupt the display info! I think merging/rebasing it to the main will get it fixed and running with networking / GPU monitoring. @gerph if you do that, please let me know, then I'll test it out as well.

phuongboss, 2023-02-20:
Confirmmed the branch `fix/replace-queued-data-send-with-immediate-plus-deferred-delay` fix the screen shuttering on macOS.

itsWobble, 2023-05-09:
I tried using the branch #60 I'm getting a COMM error "00:06:31 [ERROR] Cannot find COM port automatically, please set it manually in config.yaml".

nukdokplex, 2023-05-28:
Got this on Manjaro Linux
[image]

Here is logs:
```
мая 28 20:42:08 ndp-desktop python3[616]: 28.05.2023 20:42:08 [WARNING] Tray icon is not supported on your platform
мая 28 20:42:08 ndp-desktop python3[616]: 28.05.2023 20:42:08 [INFO] Display reset (COM port may change)...
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Auto detected COM port: /dev/ttyACM0
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [INFO] HW revision A does not support backplate LED color setting
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Image: BACKGROUND
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: CPU_MODEL
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: GPU_MODEL
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: RAM_MODEL
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: DOWNLOAD_RATE
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: UPLOAD_RATE
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: DISK_USAGE
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [INFO] Detected AMD GPU(s)
мая 28 20:50:44 ndp-desktop python3[616]: Exception in thread Queue_Handler:
мая 28 20:50:44 ndp-desktop python3[616]: Traceback (most recent call last):
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 621, in write
мая 28 20:50:44 ndp-desktop python3[616]:     n = os.write(self.fd, d)
мая 28 20:50:44 ndp-desktop python3[616]: OSError: [Errno 5] Ошибка ввода/вывода
мая 28 20:50:44 ndp-desktop python3[616]: During handling of the above exception, another exception occurred:
мая 28 20:50:44 ndp-desktop python3[616]: Traceback (most recent call last):
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
мая 28 20:50:44 ndp-desktop python3[616]:     self.run()
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/threading.py", line 953, in run
мая 28 20:50:44 ndp-desktop python3[616]:     self._target(*self._args, **self._kwargs)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/scheduler.py", line 74, in wrap
мая 28 20:50:44 ndp-desktop python3[616]:     scheduler.run()
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/sched.py", line 151, in run
мая 28 20:50:44 ndp-desktop python3[616]:     action(*argument, **kwargs)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/scheduler.py", line 64, in periodic
мая 28 20:50:44 ndp-desktop python3[616]:     action(*actionargs)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/scheduler.py", line 163, in QueueHandler
мая 28 20:50:44 ndp-desktop python3[616]:     f(*args)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/lcd/lcd_comm.py", line 99, in WriteData
мая 28 20:50:44 ndp-desktop python3[616]:     self.lcd_serial.write(bytes(byteBuffer))
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 655, in write
мая 28 20:50:44 ndp-desktop python3[616]:     raise SerialException('write failed: {}'.format(e))
мая 28 20:50:44 ndp-desktop python3[616]: serial.serialutil.SerialException: write failed: [Errno 5] Ошибка ввода/вывода
```

evilpete, 2023-10-01:
I am seeing the issue, exactly like @SaplingLTD 

MacAir  (Intel Core i5) Monterey 12.6.8

katanza12, 2023-10-30:
Same issue: MacBook Pro M1 Sonoma 14.0. I tried to remove background image, dynamic text and other graphics, put only a static text (starting with simple.program.py) but without success.

a3rosol, 2024-02-23:
I'll pile on. M1 Ultra. I had to specify the com_port manually and it didn't work with the tty device but seems to work most of the time with the cu device. you can find them using `ls /dev/{tty,cu}.*` mine shows up as `/dev/cu.usbmodemUSB35INCHIPSV21	`. But no matter which theme I chose, I always get artifacts on the screen when refreshes happen.

jeffsf, 2024-05-08:
Similar defects being seen on macOS 14.4.1 (Sonoma) and Intel in #512 

If I've found the right object, changing from RTS/CTS to DSR/DTR did not seem to change the behavior

```
    # jmk -- try adjusting serial parameters
    lcd_serial: serial.serialposix.Serial = display.lcd.lcd_serial
    from pprint import pprint
    pprint(lcd_serial.get_settings())
    lcd_serial.rtscts = False
    lcd_serial.dsrdtr = True
    pprint(lcd_serial.get_settings())

    # Initialize the display
    display.initialize_display()
```

Numbski, 2024-06-12:
Just chiming in - I am on an intel mac and getting this exact same behavior, so it isn't a CPU issue.

Numbski, 2024-06-12:
Silly question - could it be the baud rate?  Per the documentation, none of this is plain text.  We're generating bitmaps and then pushing them to the serial port.  Could be that the bitmap gets corrupted in the transfer process?

opless, 2024-06-12:
It's something to do with the mac drivers, I'm sure.

oldirtdog, 2024-12-16:
Chiming in... I'm getting the same issue on my M3 Max... I'm pretty stuck here... sounds like this might not be for Mac folks :(

[image]

gregsadetsky, 2025-01-06:
M1 Max here, macOS 12.7. Running `simple-program.py` from the latest release (3.7.0) of this software shows this:

[image]

I tried the #60 PR code but that did not work either. thanks a lot!

gregsadetsky, 2025-01-12:
hey all, I've spent a good chunk of time debugging this. it's not an easy problem (and it isn't fully fixed) but hopefully what I found can help make some progress towards fixing this.

I'm using a simplified version of [simple-program.py](https://github.com/mathoudebine/turing-smart-screen-python/blob/main/simple-program.py) to show [this image](https://github.com/mathoudebine/turing-smart-screen-python/blob/main/res/backgrounds/example_480x320.png)

I used to get a very [corrupted](https://github.com/user-attachments/assets/920e92e2-af62-4a3e-b108-585d6ae00060) image (see my comment just above). now I get this:

[image]

[image]

there are a few things to notice: although the image is "correct" horizontally speaking, vertically, its height is double of the original. looking at the image: you'll see 15 "columns" (15 * 32 = 480, the width here). the columns 0, 3, 5, etc (if we're 0-indexed) are actually on a different vertical line than columns 1, 2, 4, etc. - this is why the height is doubled.

you can see/get my code here: https://github.com/gregsadetsky/turing-smart-screen-python

I've hacked it enough that it will probably only work for showing full images (i.e. it won't work for `DisplayText` etc. calls that only draw to a portion of the screen). in any case, it's still not completely fixed, but hopefully this can shed a light.

the interesting/most hacked bits are [in serial_write](https://github.com/gregsadetsky/turing-smart-screen-python/blob/main/library/lcd/lcd_comm.py#L137).

clearly... some macos usb/serial buffering is interfering with the display's chip. as noted in my comment above, I ran simple-program.py on a raspberry pi and it worked perfectly straight out of the box. (it was frustrating to see, but I'm happy for anyone using linux!)

phucynwa, 2025-03-08:
@gregsadetsky Can I run it directly from my Mac without using VituralBox ?

mathoudebine, 2025-03-20:
Hi everyone!

I may have fixed the issue with this simple change on the serial communication:
https://github.com/mathoudebine/turing-smart-screen-python/pull/721/commits/ac169f375b20aa33421ab1ccbf7d130a5eaf0504

By manually flushing the serial buffer regularly, it has solved the corruption problem on my setup:
- MacBook Air (13-inch, 2017)
- macOS Monterrey 12.7.6

A pull request has been opened with this fix here: #721 

Before merging, I would like to have your feedback, especially for people with Apple M1 CPU.
You can use the branch `fix/7-screen-displays-corrupted-images-on-mac` or just apply the on-line fix manually.

Thanks!

mathoudebine, 2025-04-01:
Reopened because some corruptions still occur

mathoudebine, 2025-04-01:
@gregsadetsky @oldirtdog what display do you have? Is it a Turing rev. A?
Can you try with some portrait content if possible? Thanks

gregsadetsky, 2025-04-02:
It's a Rev A yeah. I've been using simple-program.py for all of my tests, should I have been using some other sample code? thanks!

tapsu01, 2025-04-05:
I get this on MacOS Ventura 13.7.5:
[image]

oldirtdog, 2025-05-13:
Tried in portrait mode... same garbage showing up on screen.

aptonline, 2026-04-27:
Can I confirm if it's just revision A displays that have this issues or is it all displays?

gregsadetsky, 2026-05-10:
@marcelors27 amazing work! Would you be able to share what code worked for you on macos?

amarok30, 2026-08-21:
**Root cause found + verified working fix on macOS** (Mac mini M4, macOS 26.5.2, Turing/UsbMonitor 3.5" rev A — `1a86:5722`, serial `USB35INCHIPSV2`).

After a systematic reverse-engineering session (solid fills, gradients, per-packet rainbow maps, barcode-tagged bands, full USB descriptor probing), the corruption mechanism is now clear:

**The panel firmware is USB-transfer-oriented, not byte-stream-oriented.**

- It reads a command from the **start of a USB transfer** and **ignores the rest of that transfer**.
- After a `DISPLAY_BITMAP` header, it **counts exactly `w*h` pixel bytes** (RGB565 LE). Any byte arriving before the rectangle is complete — including what we believe are commands — is swallowed as pixel data.
- Once the byte stream loses sync (a single lost/mis-framed byte), the panel's pixel counter is off, and **every subsequent command is eaten as pixels** → permanent corruption until power cycle. This explains why RESET/CLEAR "don't work" once the image is corrupted, and why every run produces different garbage.
- The macOS CDC ACM serial path (`/dev/cu.usbmodemUSB35INCHIPSV21`) **repackages/fragments USB transfers arbitrarily**, so commands end up glued to pixel data or split mid-command → instant desync. This matches @gregsadetsky's 64-byte packet reordering observations and why Linux/Windows/ESP32 are unaffected.

These exact semantics are documented in [@tanakamasayuki's EspUsbHost `TuringDevice.hpp`](https://github.com/tanakamasayuki/EspUsbHost/blob/main/examples/Serial/EspUsbHostDisplayTuring/TuringDevice.hpp) (ESP32 USB-host driver for this same panel): *"The panel takes a command from the start of a transfer and ignores the rest of it"* / *"The panel counts the pixels itself"*.

**Working fix (verified end-to-end):** bypass the serial device entirely — write directly to bulk OUT endpoint `0x03` via PyUSB/libusb with strict transfer discipline:

1. every command in its **own transfer** (one `endpoint.write()` per command; a header never shares a transfer with pixel data),
2. `SET_ORIENTATION` is **11 bytes** (6-byte command + orientation+100 + width/height BE), not 16,
3. pixel data in 64-byte-aligned chunks totalling **exactly** `w*h*2` bytes,
4. nothing else is written while a bitmap is in progress,
5. if the panel ever desyncs, unplug/replug the USB-C cable — RESET only reaches the panel when the stream is in sync.

With this, the panel renders solid full-screen frames back-to-back with zero desync, and the full system monitor (themes + live stats) runs correctly on macOS.

A ready-to-use backend (subclass of `LcdCommRevA`, registered as `REVISION: A_USBMAC`) is here: https://gist.github.com/amarok30/cddaa9a9818d6830a74d9332f501047e

Note: @marcelors27's one-shot bulk write worked on their unit but not on ours (identical VID/PID/serial) — there seem to be firmware batch differences; the strict transfer discipline above is what made it reliable here. Hope this helps finally close this one — happy to open a PR with the backend if there's interest.