mathoudebine/turing-smart-screen-python #8: 3,5 inch display

Befra60, 2022-02-13:
i use one intel core i7-7700 with win10. Ist this allowed with this 3,5 ich modul?
The driver is installed an i can use the program usb Monitor.
The failure is that the display never present one color screen.
I can see the with screen and never more.
The driver is installed with one faillure description   unknown device and incomplete   win10 security entry
who can help me?

mathoudebine, 2022-02-14:
Hello, are you speaking about UsbMonitor program or did you try the python script of this project?

Befra60, 2022-02-14:
Hi Matthieu,
thanks for your reaction. The problem is the driver from USBMonitor. This driver not include the safetydescription for WIN10.
usb-C to monitor not work properly.

closed 2022-02-14 by Befra60