mathoudebine/turing-smart-screen-python #12: NVMe SDD M.2 - not reading.

tangodowndaesh, 2022-05-23:
Howdy.

Not sure if u can do a support on this, but my PC setup has 4 SSD's and a main (where the system is) a NVMe SDD M.2. for any reason idk it does not send any output (nor the temperature nor the load information). I believe not matter I have secondary ssd the software detect and should show the system (boot disk). So maybe NVMe SDD M.2 it is not supported? 

Thanks for your kindly attention.

mathoudebine, 2022-05-25:
Hi, i assume you are using the Windows app USBMonitor.exe to control your screen?
Unfortunately i can't help for support with USBMonitor.exe here because the program is not open-source ☹️
This repository is an alternative Python script to do some simple operations on the screen without using USBMonitor.exe
For support on the official app you can contact the seller of your screen, i don't use it so i cannot help you with that

tangodowndaesh, 2022-05-25:
Hi yes, i'm using USBMonitor.exe :| sorry my noob question (initially I thought exe was a compilation of this python. In this case I prefer 300% open source code than an exe file. Thanks a lot for your kindly reply. 

Cheers!!

closed 2022-06-03 by mathoudebine