mathoudebine/turing-smart-screen-python #23: Windows issue with pythonw

Cyruz143, 2022-08-23:
Last one (for tonight, sorry about the spam).

If you want to setup this script to run automatically on system start, for Windows it's easiest to use the Task Scheduler. You can set this up fairly easily to run something like `C:\Users\xxx\AppData\Local\Programs\Python\Python310\python.exe "main.py"` assuming you're starting the task in the directory where main.py is.

For a version without console output (I assume how most people would want to use this). You'd use pythonw.exe. The issue with that is that it keeps opening up a CMD prompt every second then closing, It steals mouse focus too which isn't ideal. Some Googling around seems to show this is perhaps related to https://github.com/pyinstaller/pyinstaller/wiki/Recipe-subprocess. This is some vague guess work as I'm not overly familar with Python.

Also it would be nice to easily be able to turn the screen off, either by passing some args to the main.py call or having a seperate script to call, then you could run this on user logoff etc.

mathoudebine, 2022-09-13:
After investigating, the CMD window opening every second seems related to the Nvidia SMI reading, to get Nvidia GPU metrics.  
This reading is part of the GPUtil library used by this project.  
I have to check if the bug is related in GPUtil or if we don't use it properly in this project.

Cyruz143, 2022-09-13:
Thanks for looking in to it. Really appreciate all the work you've been putting in to all this 👍

mathoudebine, 2022-09-21:
Other issues with GPUtil like #53 shows the need to use a more generic solution for HW access like #52

Cyruz143, 2022-12-10:
Good news, this seems to work now, no pop up CMD window trying to do bits!

I'll try and update the wiki, still need a nice solution to kill it when the computer turns off though to disable the screen as many motherboards still power the USB ports with the machine off. This means the screen keeps a frozen image on there. I'll poke around this in the week and try and find a nice solution.

For the curious, you can make a scheduled task in windows with the following details:

[image]

[image]

[image]

mathoudebine, 2022-12-10:
Thanks @Cyruz143!
I added your startup procedure in the wiki: https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-%3A-how-to-start