mathoudebine/turing-smart-screen-python #29: Alternative implementation as object-oriented interface (with MIT license)

gerph, 2022-08-28:
**Is your feature request related to a problem? If so, please describe the problem.**  

The problem I had was that this repo is licensed under a restrictive license that precludes my using it in my project. Additionally, I wanted to provide support for the two variants using object oriented interfaces - with the version B and version A sharing common interfaces and code.

I also wanted to be able to add the ability to invert the content easily - something I now see is also possible in the hardware A implementation.

So I wrote a new library for myself, using the information - but none of the code - from this project.

This can be found here:

https://github.com/gerph/turing-smart-screen-python-mit

**Describe the feature / solution to your problem you'd like**  

As I needed this library for myself in my project, I had to re-write it from scratch, but it might as well be available to anyone. Some of what I've done in my version is almost certainly reusable within this project - and because it's MIT licensed, that's quite compatible with integration into GPL projects such as this.

What does the my repo offer that this code doesn't?

Not that much really.

* Primarily it was written so that I didn't have to use GPL code, and to be Python 2/3 compatible.
* The class based implementation means that the two variants can have the same interface and be descended from the same base class.
* This means that hardware features not provided can be implemented in software, or report errors.
* The implementation also provides for inverting the rendering of the bitmap (in software), ie flipping the image horizontally or vertically.
* The sleep-between-bitmap-data has been reworked so that the sleeping happens only when necessary when the next command happens, rather than being forced at the end of a bitmap update. This should be more efficient as it allows the python to get on with other things rather than stalling, and only sleeps when it's absolutely necessary.
* And there's a constructor function that uses the fact that the old interfaces don't respond to HELLO to select the correct objects.

Take the code and use it if you want. Take the ideas. It's freely given under MIT license.

I've tested my code with the example functions in this repository and it appears to work quite well with my library switched in.

**Describe alternatives you've considered / and or tested**  

I briefly considered requesting the change to the license from GPL, but people that select GPL are commonly averse to changing away from it, and with multiple contributors who might need to agree to such a change, it wasn't worth it. Easier to just re-write it.

**Screenshots / photos & mockups of the Turing screen**  

Not really screenshots, but the example code that I've been using from this project (which is not part of the MIT version) can be used with the new library. Readlly the only bits that were changed were the functions in the test code which performed operations - they were just updated to call the object instead of doing the work themselves.

```python
#!/usr/bin/env python3
# A simple Python manager for "Turing Smart Screen" 3.5" IPS USB-C display
# https://github.com/mathoudebine/turing-smart-screen-python

import os
import signal
import struct
from datetime import datetime
from time import sleep

import serial  # Install pyserial : pip install pyserial
from PIL import Image, ImageDraw, ImageFont  # Install PIL or Pillow

# Set your COM port e.g. COM3 for Windows, /dev/ttyACM0 for Linux...
COM_PORT = "/dev/ttyACM0"
# COM_PORT = "COM5"
# MacOS COM port:
COM_PORT = '/dev/cu.usbmodem2017_2_251'

# The new device has a serial number of '2017-2-25'

import turing_smart_screen

def Clear(display):
    # Unknown what command this is
    print("Clear display")
    display.clear()

def Orientation(display, state: int):
    print("Orientation: %r" % (state,))
[... 165 lines cut ...]

        DisplayProgressBar(lcd_comm, 10, 40,
                           width=140, height=30,
                           min_value=0, max_value=100, value=bar_value,
                           bar_color=(255, 255, 0), bar_outline=True,
                           background_image="res/example.png")

        DisplayProgressBar(lcd_comm, 160, 40,
                           width=140, height=30,
                           min_value=0, max_value=19, value=bar_value % 20,
                           bar_color=(0, 255, 0), bar_outline=False,
                           background_image="res/example.png")

        bar_value = (bar_value + 2) % 101

    SetLighting(lcd_comm, 0, 0, 0)

    ScreenOff(lcd_comm)

    ser.close()
```

**Additional context**  

Thanks for the library, and I might try to integrate the object interface into the system monitor branch later, as it looks like that's going the object oriented route which might benefit from that interface.

mathoudebine, 2022-08-28:
Hi and sorry for the license inconvenience.

The original reason of choosing GPLv3 was to avoid people making and distributing closed source versions of this repository. I do not want the product manufacturer and/or seller to package and distribute this as if it was their original software.
Sorry to learn that you needed to re-implement everything from scratch.

As you have seen, this project will soon switch to an object-oriented implementation with the 2.0 release, so i will not maintain the current release anymore. Thanks for the example code though.
I pinned this issue in case other people have issues with the GPL license.

gerph, 2022-08-28:
That's fine - I really don't mind people using GPL if they believe that they want to keep things open, obviously. I also don't want people to distribute things as their own, but I feel that the MIT license conveys that - they're still required to retain attribution, but they aren't forced to bring their entire product under the terms of another license.

In this case, I love that you'd done a load of work on this to get it to this point and that it's available. I can't use it, but I can still contribute and help to make it better. And I love that you've been so responsive with it too - *that* is more important to open source :-)

closed 2022-09-13 by mathoudebine

fakerybakery, 2024-03-14:
Just found this repo - super cool @mathoudebine! Have you considered using the zlib license, which doesn't place restrictions on linking but prohibits people from passing it off as their own? Thanks!