mathoudebine/turing-smart-screen-python #32: Add date/time metrics

mathoudebine, 2022-08-30:
**Is your feature request related to a problem? If so, please describe the problem.**  
There is no entries in theme.yaml for date/time

**Describe the feature / solution to your problem you'd like**  
Entries in theme.yaml to be able to display date and time on screen, with configurable format

**Describe alternatives you've considered / and or tested**  
Using static_text entries but it is not updated at runtime

**Environment:**  
 - Release 2.0.0-beta.1, `feature/system-monitor` branch
 - All OS with version [e.g. Windows 11, Ubuntu 22.04]
 - Python version > 3.7

waynehuang85, 2022-09-11:
In addition to the date/time metrics, would love to see the Network In/Out if readily availalbe. 

Why - this secondary monitor screen will be a nice addition to NAS system like Synology DS or other headless home theater PC(HTPC)

Currently using it on my Raspberry Pi 3B+ (running Retropie) however CPU temp is not supported

Nealtron, 2022-09-18:
> In addition to the date/time metrics, would love to see the Network In/Out if readily availalbe.
>
> Why - this secondary monitor screen will be a nice addition to NAS system like Synology DS or other [...]

I love this idea, but it should probably be it's own feature request.

mathoudebine, 2022-09-19:
@Nealtron and @waynehuang85 you can subscribe to this feature request for network metrics: https://github.com/mathoudebine/turing-smart-screen-python/issues/51

Rollbacke, 2022-10-20:
Hi,

**My purpose:**

I used datetime.

In `main.py`, add:
```python
scheduler.DateStats()
```

In `scheduler.py`, add:
```python
@async_job("Date_Stats")
@schedule(timedelta(seconds=THEME_DATA['STATS']['DATE'].get("INTERVAL", None)).total_seconds())
def DateStats():
    # logger.debug("Refresh date stats")
    stats.Date.stats()
```

In `stats.py`, add:
```python
class Date:
    @staticmethod
    def stats():
        date_now = datetime.datetime.now()

        if THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("SHOW", False):
            display.lcd.DisplayText(
                text=f"{date_now.day}-{date_now.month}-{date_now.year}",
                x=THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("X", 0),
                y=THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("Y", 0),
                font=THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("FONT", "roboto-mono/RobotoMono-Regular.ttf"),
                font_size=THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("FONT_SIZE", 10),
                font_color=THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("FONT_COLOR", (0, 0, 0)),
                background_color=THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("BACKGROUND_COLOR", (255, 255, 255)),
                background_image=get_full_path(THEME_DATA['PATH'],
                                               THEME_DATA['STATS']['DATE']['DAY']['TEXT'].get("BACKGROUND_IMAGE",
                                                                                               None))
            )

        if THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("SHOW", False):
            display.lcd.DisplayText(
                text=f"{date_now.strftime('%H:%M:%S')}",
                x=THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("X", 0),
                y=THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("Y", 0),
                font=THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("FONT", "roboto-mono/RobotoMono-Regular.ttf"),
                font_size=THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("FONT_SIZE", 10),
                font_color=THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("FONT_COLOR", (0, 0, 0)),
                background_color=THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("BACKGROUND_COLOR", (255, 255, 255)),
                background_image=get_full_path(THEME_DATA['PATH'],
                                               THEME_DATA['STATS']['DATE']['HOUR']['TEXT'].get("BACKGROUND_IMAGE",
                                                                                                None))
            )
```

In your `theme.yaml`:
```python
  DATE:
    # In seconds. Longer intervals cause this to refresh more slowly.
    # Setting to lower values will display near real time data,
    # but may cause significant CPU usage or the display not to update properly
    INTERVAL: 1
    DAY:
      TEXT:
        SHOW: True
        X: 4
        Y: 4
        FONT: roboto-mono/RobotoMono-Regular.ttf
        FONT_SIZE: 16
        FONT_COLOR: 53, 191, 92
        # BACKGROUND_COLOR: 0, 0, 0
        BACKGROUND_IMAGE: background.png
    HOUR:
      TEXT:
        SHOW: True
        X: 160
        Y: 4
        FONT: roboto-mono/RobotoMono-Regular.ttf
        FONT_SIZE: 16
        FONT_COLOR: 53, 191, 92
        # BACKGROUND_COLOR: 0, 0, 0
        BACKGROUND_IMAGE: background.png
```

RussNelson, 2022-10-26:
I created a pull request for it.

Rollbacke, 2022-10-30:
Thanks @RussNelson ;)

mathoudebine, 2022-11-06:
Implemented in #72

closed 2022-11-06 by mathoudebine