mathoudebine/turing-smart-screen-python #40: Installation Problem

T0mGates11, 2022-09-09:
Hey, firts of all you got a command line in step 3
Should be: python -m pip install -r requirements.txt 

Second i got a problem while installing and i cant figure it out. I installed all modules. i checked it.
I followed all the steps.
if i want to open main.py this happens: 
C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1>python main.py
20:14:42 [INFO] Loading theme 3.5inchTheme2 from res/themes/3.5inchTheme2/theme.yaml
Traceback (most recent call last):
  File "C:\Users\rigam\AppData\Local\Programs\Python\Python310\lib\site-packages\pyadl\adl_api.py", line 52, in <module>
    _libadl = CDLL("atiadlxx.dll")
  File "C:\Users\rigam\AppData\Local\Programs\Python\Python310\lib\ctypes\__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'atiadlxx.dll' (or one of its dependencies). Try using the full path with constructor syntax.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\rigam\AppData\Local\Programs\Python\Python310\lib\site-packages\pyadl\pyadl.py", line 26, in <module>
    from .adl_api import *
  File "C:\Users\rigam\AppData\Local\Programs\Python\Python310\lib\site-packages\pyadl\adl_api.py", line 55, in <module>
    _libadl = CDLL("atiadlxy.dll")
  File "C:\Users\rigam\AppData\Local\Programs\Python\Python310\lib\ctypes\__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'atiadlxy.dll' (or one of its dependencies). Try using the full path with constructor syntax.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1\main.py", line 19, in <module>
    import library.scheduler as scheduler
  File "C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1\library\scheduler.py", line 8, in <module>
    import library.stats as stats
  File "C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1\library\stats.py", line 12, in <module>
    import pyadl
  File "C:\Users\rigam\AppData\Local\Programs\Python\Python310\lib\site-packages\pyadl\__init__.py", line 11, in <module>
    from .pyadl import *
  File "C:\Users\rigam\AppData\Local\Programs\Python\Python310\lib\site-packages\pyadl\pyadl.py", line 28, in <module>
    raise ADLError("Driver not found!")
pyadl.pyadl.ADLError: Driver not found!

Any ideas? 
Thanks

mathoudebine, 2022-09-10:
Hello,
Thanks for raising this issue, I will fix the documentation for the requirements install.
For the runtime bug, can you provide the following info to help:

 - Release or branch of this project [e.g. 1.1.1, 2.0.1, `main` branch...]
 - OS with version [e.g. Windows 11, Ubuntu 22.04]
 - Python version [e.g. Python 3.8, to get it run `python3 --version`]
 - Hardware [e.g. Intel CPU, Nvidia GPU, Raspberry Pi 3 B+ ...]

Thanks

T0mGates11, 2022-09-10:
Hey, thanks for the quick response. I used the newest main branch
Im using Windows 11 Pro, python 3.10 
CPU: i7-9700K, GPU: RTX 2080 Super
RAM: 32GB Corsair Vengeance
And im using the Flagship version

T0mGates11, 2022-09-10:
While the simple_program worked flawlessly, the main.py does not work. 
Above you saw the error log, after some investigation me and my friend who knows a bit of python replaced line 11 to 14 in library/stats.py
to "pyadl = None". We figured that, it wrongly imported pyadl and tried to use it even though I do not use an AMD GPU.
Wherever it was good or not now main.py could not recognise the device:

C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1>python main.py
21:04:19 [INFO] Loading theme 3.5inchTheme2 from res/themes/3.5inchTheme2/theme.yaml
21:04:20 [DEBUG] Static COM port: COM4
21:04:21 [WARNING] Device not recognised (short response to HELLO)
Traceback (most recent call last):
  File "C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1\main.py", line 56, in <module>
    display.initialize_display()
  File "C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1\library\display.py", line 57, in initialize_display
    self.lcd.InitializeComm()
  File "C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1\library\lcd_comm_rev_b.py", line 117, in InitializeComm
    self.Hello()
  File "C:\Users\rigam\Desktop\turing-smart-screen-python-2.0.1\library\lcd_comm_rev_b.py", line 95, in Hello
    if response[0] != Command.HELLO or response[-1] != Command.HELLO:
IndexError: index out of range

mathoudebine, 2022-09-10:
Thanks for investigating, you're right about the issue with pyadl that should not be loaded if you do not have an AMD GPU. I will fix that in a next version soon.
For the second error i've never seen it, it seems not related to the change you made. Maybe the screen got stuck between 2 commands, can you unplug it then plug it back in to reset it then try again?

T0mGates11, 2022-09-10:
Thanks,
i did what you said, sadly with the same result.

Im looking forward to the new Version, hopefully it will work then.
Thanks for your work.

mathoudebine, 2022-09-12:
Bug now fixed in release 2.0.2 and successfully tested on Windows 11 with Nvidia GPU
@T0mGates11 can you check if it fixed your issue? Thanks!

closed 2022-09-12 by mathoudebine