mathoudebine/turing-smart-screen-python #45: Windows 10, laptop

bloor, 2022-09-15:
**Describe the bug**  
Error when trying to execute the script from CMD:
`serial.serialutil.PortNotOpenError: Attempting to use a port that is not open`

**To Reproduce**  
Download the repository
Installed all the py modules required

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Release or branch of this project [e.g. 1.1.1, 2.0.0-alpha4, `feature/system-monitor` branch]
 - Windows 10 (latest update to date)
 - Python version 3.9.1
 - Intel CPU 10875H, Nvidia 2070 RTX 

**Additional context**  
I am plugging the USB directly in the laptop, not in the usb-HUB

mathoudebine, 2022-09-15:
Hi,
I can confirm that on some Windows computers, the COM port auto-discovery does not work.
I can reproduce it on at least one computer, also a laptop with Windows 10.
Until I investigate this issue, you can edit `config.yaml` and indicate your COM port manually instead of `AUTO`

mathoudebine, 2022-09-15:
And it is normal for the LCD to stay off when plugged if you have a B or flasghip revision.
The LCD should turn on after you run the program (without errors).

bloor, 2022-09-15:
Yes. it's variant B. Ok.. i was just a bit worried it might be dead or something. 
I already tried with that software, but it does not run even after adding some "missing dlls". + its full chinese the files might be corrupted due to charset or something.
I`ll check some workarounds.

bloor, 2022-09-15:
Manually putting COM3 worked, but I also didn't noticed my REVISION was A in config, instead of B.
WORKS AS INTENDED 👍 freaking love it. 10x for the help

bloor, 2022-09-15:
donzo

closed 2022-09-15 by bloor