mathoudebine/turing-smart-screen-python #51: Add network metrics

mathoudebine, 2022-09-19:
**Is your feature request related to a problem? If so, please describe the problem.**  
There is no entries in theme.yaml for network metrics (upload/download speed)

**Describe the feature / solution to your problem you'd like**  
Entries in theme.yaml "STATS" to be able to display network metrics

Rollbacke, 2022-10-20:
I tried and it work, but the network card name is a problem.
I used psutil and a temporary file.

**My example :**

In `main.py`, add:
```python
scheduler.NetStats()
```

In `scheduler.py`, add:
```python
@async_job("Net_Stats")
@schedule(timedelta(seconds=THEME_DATA['STATS']['NET'].get("INTERVAL", None)).total_seconds())
def NetStats():
    # logger.debug("Refresh net stats")
    stats.Net.stats()
```

In `stats.py`, add:
```python
class Net:
    @staticmethod
    def stats():
        tot_before_file = "/tmp/tot_before"
        pnic_before_file = "/tmp/pnic_before"

        tot_after = psutil.net_io_counters()
        pnic_after = psutil.net_io_counters(pernic=True)
        if os.path.exists(tot_before_file) and os.path.isfile(tot_before_file):
            with open(tot_before_file, "rb") as file:
                tot_before = pickle.load(file)
        else:
            tot_before = tot_after
        if os.path.exists(pnic_before_file) and os.path.isfile(pnic_before_file):
            with open(pnic_before_file, "rb") as file:
                pnic_before = pickle.load(file)
        else:
            pnic_before = pnic_after

        with open(tot_before_file, "wb") as file:
            pickle.dump(tot_after, file)
        with open(pnic_before_file, "wb") as file:
            pickle.dump(pnic_after, file)

        upload_wlo1 = pnic_after["wlo1"].bytes_sent - pnic_before["wlo1"].bytes_sent
        uploaded_wlo1 = pnic_after["wlo1"].bytes_sent
        download_wlo1 = pnic_after["wlo1"].bytes_recv - pnic_before["wlo1"].bytes_recv
        downloaded_wlo1 = pnic_after["wlo1"].bytes_recv
        upload_eth0 = pnic_after["eth0"].bytes_sent - pnic_before["eth0"].bytes_sent
        uploaded_eth0 = pnic_after["eth0"].bytes_sent
        download_eth0 = pnic_after["eth0"].bytes_recv - pnic_before["eth0"].bytes_recv
        downloaded_eth0 = pnic_after["eth0"].bytes_recv

        if THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("SHOW", False):
            display.lcd.DisplayText(
                text=f"{bytes2human(upload_wlo1):>5} B/s",
                x=THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("X", 0),
                y=THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("Y", 0),
                font=THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("FONT", "roboto-mono/RobotoMono-Regular.ttf"),
                font_size=THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("FONT_SIZE", 10),
                font_color=THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("FONT_COLOR", (0, 0, 0)),
                background_color=THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("BACKGROUND_COLOR", (255, 255, 255)),
                background_image=get_full_path(THEME_DATA['PATH'],
                                               THEME_DATA['STATS']['NET']['WLO1']['UPLOAD']['TEXT'].get("BACKGROUND_IMAGE",
                                                                                               None))
            )

        if THEME_DATA['STATS']['NET']['WLO1']['UPLOADED']['TEXT'].get("SHOW", False):
            display.lcd.DisplayText(
[... 162 lines cut ...]
          BACKGROUND_IMAGE: background.png
      DOWNLOAD:
        TEXT:
          SHOW: True
          X: 254
          Y: 311
          FONT: roboto-mono/RobotoMono-Regular.ttf
          FONT_SIZE: 12
          FONT_COLOR: 53, 191, 92
          # BACKGROUND_COLOR: 0, 0, 0
          BACKGROUND_IMAGE: background.png
      DOWNLOADED:
        TEXT:
          SHOW: True
          X: 192
          Y: 311
          FONT: roboto-mono/RobotoMono-Regular.ttf
          FONT_SIZE: 12
          FONT_COLOR: 53, 191, 92
          # BACKGROUND_COLOR: 0, 0, 0
          BACKGROUND_IMAGE: background.png
```

But I have one last problem. Depending on the bit rate, if there is one more or less digit, it leaves "s" behind (see photo).
[image]

RussNelson, 2022-10-26:
@Rollbacke Suggestion: use pnic_after.keys() to get the interfaces. If the user hasn't set something in the Theme Data that matches one of these interfaces, throw a traceback:
for dev in THEME_DATA['NET']['NICS']:
    assert dev in pnic_after.keys(), "You have specified a network interface that cannot be found in theme.yaml."

mathoudebine, 2022-10-26:
Thanks @Rollbacke for your work!
Do you want to open a pull request to integrate it to this project?

RussNelson, 2022-10-26:
I created a pull request for @Rollbacke 's work.

Rollbacke, 2022-10-30:
Yeah, sorry, I wanted to look at how to do PR (I've never done one), but I didn't have time...

RussNelson, 2022-10-30:
It's pretty easy.
(on the git page for the repo)
git clone your-fork-url
git checkout -b newbranchname
make your changes
git commit -a -m "why you made changes"
git push
That will print a command you need to force the origin. Copy/paste it.
Then go to your github page for the repo.
You'll be offered a "click here to create a PR"
Click it, and put in whatever notes you want to go with the PR. In my
group, we tell the testers how to test the branch before they commit it to
main.

Rollbacke, 2022-11-02:
Hi team.
@RussNelson, Can you cancel your two PR (network metrics and date)?
I will take care of adding 2/3 things to the code:
- integrate it into themes
- check if the network cards exist, otherwise put "0"

I will propose the addition of my theme of which here is a preview
[image]

RussNelson, 2022-11-02:
No need for that. Just check out the associated branch and add your commits to it. They'll show up in the PR and everything copacetic.

mathoudebine, 2022-11-06:
Implemented in #72

closed 2022-11-06 by mathoudebine