mathoudebine/turing-smart-screen-python #52: Open Hardware Monitor integration

Cyruz143, 2022-09-20:
**Is your feature request related to a problem? If so, please describe the problem.**  
In theory this _should_ hopefully resolve a few issues.
- Dynamic CPU clock speed under Windows
- CPU temperature under Windows
- Hopefully allow modern AMD gpu stats fixing https://github.com/mathoudebine/turing-smart-screen-python/issues/49

**Describe the feature / solution to your problem you'd like**  
Integration of the Open Hardware Monitor (https://github.com/openhardwaremonitor/openhardwaremonitor) libs to replace various other limited 3rd party Python libs for Windows.

**Describe alternatives you've considered / and or tested**  
I'm sure you were already aware of this, just thought it might save people making multiple issues requesting features/reporting bugs that are now know limitations of the currently used libs.

**Additional considerations**  
I hope this fits your license seeing as OHM is open source, not sure if you want to include a .dll compiled by someone else.

mathoudebine, 2022-09-21:
Very good idea indeed!
I was also thinking about [HWiNFO SDK](https://www.hwinfo.com/sdk/) but it is not open-source, so i like Open Hardware Monitor better
I plan to work on it soon!

Cyruz143, 2022-09-21:
> Very good idea indeed! I was also thinking about [HWiNFO SDK](https://www.hwinfo.com/sdk/) but it is not open-source, so i like Open Hardware Monitor better I plan to work on it soon!

Your choice might still be better. OHM is looking fairly quiet on Github, lots of issues not many actual commits going on...

mathoudebine, 2022-09-21:
I found https://github.com/LibreHardwareMonitor/LibreHardwareMonitor a fork of Open Hardware Monitor still maintained, will give it a try

mathoudebine, 2022-11-14:
Hi guys, I've started the development of the LibreHardwareMonitor integration for Windows platform. This will not be a quick task though.

I already found 3 ways:
1. the preferred way: call the LibreHardwareMonitorLib.dll library directly from Python like  https://github.com/gerfra/Temperature-Hardware-LibreHardwareMonitorLib
2. use the WMI callbacks of the LibreHardwareMonitor.exe program like https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitor/TestScripts/basicwmi.py 
3. use the RestAPI of the LibreHardwareMonitor.exe program like https://github.com/Gokender/minotorr

mathoudebine, 2022-12-09:
Hello @Cyruz143
The first beta release that supports LibreHardwareMonitor has been released!
https://github.com/mathoudebine/turing-smart-screen-python/releases/tag/2.2.0-beta.1
If you have time to test it, i would be happy to have your feedback!

Cyruz143, 2022-12-09:
Awesome work! Will give it a run this weekend and let you know.

Cyruz143, 2022-12-12:
Ran it this weekend, no real issues to report, worked as expected! I did see the CPU effective clock being incorrect but that's been picked up in another ticket already with a suggested fix.

closed 2022-12-13 by mathoudebine