mathoudebine/turing-smart-screen-python #53: Monitoring causes GPU stuttering

Cyruz143, 2022-09-21:
**Describe the bug**  
Running the main.py causes a constant micro stutter on GPU output, measurablable with tools and noticeable in games (frametimes spike).

**To Reproduce**  
Steps to reproduce the behavior:
1. Open a browser page to https://www.testufo.com/animation-time-graph
2. Allow the test to settle
3. Run main.py

**Expected behavior**  
Monitoring shouldn't impact performance

**Gif**
[image]

**Environment:**  
 - Release 2.0.3
 - Windows 10
 - Python 3.10.6
 - AMD 5900x / Nvidia 3080

**Additional context**  
I'd like to see if other people can reproduce this as things like this could have varying degrees of impact depending on a bunch of variables.

I ran Open Hardware Monitor to see if that impacts the GPU the same way and it doesn't so it's either in the gputil lib or way Python calls it?

Cyruz143, 2022-09-21:
I can replicate this with a quick and dirty test file

```
import time
import GPUtil

while True:
    GPUtil.showUtilization()
    time.sleep(1)
```

mathoudebine, 2022-09-21:
Thanks for investigating, issue seems linked to GPUtil though i still need to check behavior with pyadl (AMD GPU)
Anyway, the need to integrate OpenHardwareMonitor or alternative would be a better solution, i plan to work on it as soon as i can.
Issue would still be present on Linux though. If you have time you can also open this issue on [gputil github](https://github.com/anderskm/gputil) and link it here.

Cyruz143, 2022-09-21:
>If you have time you can also open this issue on [gputil github](https://github.com/anderskm/gputil) and link it here.

I was going to but it's not had much activity in the last few years so I'm not sure if it's still maintained? I'll make one and link it across.

closed 2022-12-13 by mathoudebine

mathoudebine, 2022-12-13:
Fixed in the 2.2.0 release for Windows platforms