mathoudebine/turing-smart-screen-python #59: Access denied error when running program

BishDub, 2022-09-30:
**Describe the bug**  
Access denied error when running program

**To Reproduce**  
Steps to reproduce the behavior:
1. Open shell as admin, 
2. run simple-program.py

**Expected behavior**  
Program loads

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the Turing screen to help explain your problem.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Release or branch of this project: 2.0.3
 - OS with version: Windows 10 21H2
 - Python version: 3.10.7
 - Hardware: AMD 3700x, Nvidia 3070s

**Additional context**  
Turing model A

Error message:

    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM7': PermissionError(13, 'Access is denied.', None, 5)

mathoudebine, 2022-10-09:
Hello, there is no need to run the program as administrator.
First thing you can check is that COM7 is the correct serial port for your screen. If not, you can replace "AUTO" here https://github.com/mathoudebine/turing-smart-screen-python/blob/main/simple-program.py#L19 by the correct COM port in case the auto-discovery failed.
Second thing you can check is that the vendor program UsbMonitor.exe is not running in background.

BishDub, 2022-10-09:
Thanks for the reply. I've tried manually setting the com port, i also tried changing the com port. UsbMonitor isn't running either.

moldark-debug, 2022-11-01:
I have the same issue. No other programm uses the COM port. Changing the COM port, for example to COM6, does not help. I've tried changing from auto-discovery to a fixed COM port, but issue is the same.

File "C:\Users\test\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Zugriff verweigert', None, 5)

Python 3.10.8
Windows 10 21H1
Build 19043.2130

BishDub, 2022-11-02:
I ended up using Sim Hub as an alternative

mathoudebine, 2022-11-03:
Hi @moldark-debug
There is indeed a bug in auto-discovery, but if you set the COM port manually it should have been running fine
Can you send the complete log when you run the program?
Are you sure you entered the correct COM port for the screen in `config.yaml` file?

moldark-debug, 2022-11-03:
> Hi @moldark-debug There is indeed a bug in auto-discovery, but if you set the COM port manually it should have been running fine Can you send the complete log when you run the program? Are you sure you entered the correct COM port for the screen in `config.yaml` file?

I entered the correct COM port in the config.yaml file and get the same error:

00:39:54 [INFO] Loading theme 3.5inchTheme2 from res/themes/3.5inchTheme2/theme.yaml
00:39:55 [DEBUG] Static COM port: COM3
00:39:55 [INFO] Display reset (COM port may change)...
00:39:56 [DEBUG] Static COM port: COM3
Traceback (most recent call last):
  File "F:\turing-smart-screen-python-2.0.3\main.py", line 56, in <module>
    display.initialize_display()
  File "F:\turing-smart-screen-python-2.0.3\library\display.py", line 54, in initialize_display
    self.lcd.Reset()
  File "F:\turing-smart-screen-python-2.0.3\library\lcd_comm_rev_a.py", line 68, in Reset
    self.openSerial()
  File "F:\turing-smart-screen-python-2.0.3\library\lcd_comm.py", line 63, in openSerial
    self.lcd_serial = serial.Serial(lcd_com_port, 115200, timeout=1, rtscts=1)
  File "C:\Users\testuser\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\testuser\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Users\testuser\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Zugriff verweigert', None, 5)

fretshot, 2022-11-10:
Make sure UsbMonitor isn't running.

mathoudebine, 2023-05-24:
Closing this bug as inactive since November, tag me or open another bug report if you still have issues

closed 2023-05-24 by mathoudebine