mathoudebine/turing-smart-screen-python #63: Fourth hardware revision identifies as A

RussNelson, 2022-10-26:
The TSS I just received identifies as A from the software end, but has a fully enclosed front and back. No electronics visible other than the two USB-C connectors.

https://photos.app.goo.gl/ki2Gb5mvS6evMk186

RussNelson, 2022-10-26:
[image]

RussNelson, 2022-10-26:
I've added it to the AllVersions.jpg picture in res/docs/

[image]

RussNelson, 2022-10-26:
Submitted a PR instead.

closed 2022-10-26 by RussNelson

mathoudebine, 2022-10-26:
Interesting! They finally made a proper case for the A revision
It means they also continue to sell it for which i'm surprised. Can you tell me where you bought it?
Don't hesitate to open bugs if you see something wrong when using this project with your screen!

RussNelson, 2022-10-26:
https://www.geekbuying.com/item/3-5-Inch-IPS-TYPE-C-Secondary-Screen-CPU-GPU-RAM-HDD-Monitoring-501430.html
Paid $20 (discounted) for two of them, but the price is now $23.