mathoudebine/turing-smart-screen-python #66: Every theme needs to know about every value

RussNelson, 2022-10-26:
**Describe the bug**  
When somebody creates a new measured value, every theme.yaml needs to specify where it should go. If you only want to display one value, you have to specify all the other values as well.

**To Reproduce**  
Apply to DateTime PR.
Config it.
Try to run it.

**Expected behavior**  
If a theme does not mention a measured value, the code should ignore that value for that theme.

mathoudebine, 2022-10-28:
Hi @RussNelson, that would be a great change indeed!
I will try to work on it next week. Basically, some try/except in the code should do the job

dabeer, 2023-10-04:
Can this issue be expanded to "generalized error handling". For example, I set the x,y of a radial sensor to 0,0, and it threw a non-recoverable exception.