mathoudebine/turing-smart-screen-python #68: 5" version

MrFr1day, 2022-10-27:
Hi,

I've a couple of questions that hopefully someone can answer:

1- I've found 5 inch versions in AliExpress that seems to use the same "UsbMonitor" software, did anyone try them? will they work if I update the resolution in the config file?

2- In one of your posts, you mentioned that the refresh rate is very slow, is this a Python limitation?
I ask because I saw a preview of the monitor in AliExpress playing an animated background with a relatively high refresh rate, although it might be using the SD card for that.

Thanks.

mathoudebine, 2022-10-28:
Hello, we had a discussion on 5" version here:
https://github.com/mathoudebine/turing-smart-screen-python/discussions/36
It seems to use UsbMonitorL.exe rather than UsbMonitor.exe, so i don't know if it is the same protocol or not.
The animated background is indeed played from the SD Card, because the serial connection through USB is too slow to do that. It also means there are specific screen commands to load content from SD card, which are not supported by this project for now.
Keep us informed if you buy it and test it with this project!
Can you post the Aliexpress link so i can take a look?
Thanks

MrFr1day, 2022-10-28:
Hi,

- I've downloaded the software of one these monitors, and it had the "L" edition of UsbMonitor.

- As for the refresh rate, will the "simulated LCD" be a realistic measure for it?
I'll not be viewing videos or have an animated BG, but I'll have alot of numbers (with transparent background), and bars on it, so I wanted to see how it compares to Logitech G15 ...

- Here are a couple of vendors that I'll probably end up buying from:
https://www.aliexpress.com/item/1005003259563629.html
https://www.aliexpress.com/item/1005004495210303.html 
Can you please take a look and tell me if you see any red flags before I order one?

Thanks.

MrFr1day, 2022-10-28:
BTW, I found one that claims it support other software (like AIDA64):

> Software support: The software supports the latest version of for AIDA64. For AIDA64 is fully compatible with under Windows, and Lcd4 Linux software package is fully compatible with under Linux. It supports for NAS / openwrt router / raspberry pi system, and car game simulator.

Will this this be easier to control?

https://www.aliexpress.com/item/1005004862895332.html

mathoudebine, 2022-11-03:
>     * As for the refresh rate, will the "simulated LCD" be a realistic measure for it?
>       I'll not be viewing videos or have an animated BG, but I'll have alot of numbers (with transparent background), and bars on it, so I wanted to see how it compares to Logitech G15 ...

The simulated LCD is not realistic at all for the refresh rate. Indeed, the most time-consuming step is due to the data transmission to the "real" LCD using serial over USB. This step is not necessary in simulated mode where a picture is generated directly on your computer.
Simulated LCD is mostly used for themes designing/debugging

>     * Here are a couple of vendors that I'll probably end up buying from:
>       https://www.aliexpress.com/item/1005003259563629.html
>       https://www.aliexpress.com/item/1005004495210303.html
>       Can you please take a look and tell me if you see any red flags before I order one?

Both products seems fine, just to confirm that the 5" version has not been tested with this project, so it may not work at all, or only some features (e.g. no video from SD card)

> BTW, I found one that claims it support other software (like AIDA64):
>
>
> Will this this be easier to control?
>
> https://www.aliexpress.com/item/1005004862895332.html

For this product, i don't know how the LCD is controlled by a computer. There is no mention of UsbMonitor program, so it is quite sure it won't work with this project.