mathoudebine/turing-smart-screen-python #69: [DEBUG] Auto detected COM port: None

JoBiclou, 2022-10-30:
Hi,
Newbies on IT.

Trying to work with this product : turing smart screen flagship.
Work on debian 11.5, full amd...
Well,
When i run "python3 main.py" the system return me this message "[DEBUG] Auto detected COM port: None" since the second line.
And at the end, this message : "serial.serialutil.PortNotOpenError: Attempting to use a port that is not open"
Think i need to open/manage something but i don't know what and how.

Could you, please, help me.
Thanks.
Kind regards.

JoBiclou, 2022-10-30:
For more information,
After tests, seems on debian use /dev/ttyS0.
But have now issue with  this :
16:25:35 [INFO] Loading theme Terminal from res/themes/Terminal/theme.yaml
16:25:35 [DEBUG] Static COM port: /dev/ttyS0
16:25:35 [INFO] Display reset (COM port may change)...
16:25:36 [DEBUG] Static COM port: /dev/ttyS0
16:25:36 [INFO] HW revision A does not support backplate LED color setting
16:25:36 [DEBUG] Drawing Image: BACKGROUND
16:25:36 [DEBUG] Drawing Text: TEXT_EXAMPLE
16:25:36 [INFO] Detected AMD GPU(s)
the system is waiting for more than 22 minutes.

mathoudebine, 2022-11-03:
Hello, if you have the Turing Smart Screen revision Flagship you need to change the `REVISION` from `config.yaml` file. Here i see in your logs that you are using the Revision A but you need to indicate B for the flagship version. See comment directly in `config.yaml`

mathoudebine, 2022-11-03:
Please also have a look at the Wiki especially this How To Start page for more info: https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-%3A-how-to-start

JoBiclou, 2022-11-04:
Hi,
Thanks for this informations.
the "simple-program.py" return this :

# python3 simple-program.py
Selected Hardware Revision B (XuanFang screen version B / flagship)
13:38:10 [DEBUG] Auto detected COM port: None
Traceback (most recent call last):
  File "/home/easy/turing-smart-screen-python-2.0.3/simple-program.py", line 68, in <module>
    lcd_comm.InitializeComm()
  File "/home/easy/turing-smart-screen-python-2.0.3/library/lcd_comm_rev_b.py", line 117, in InitializeComm
    self.Hello()
  File "/home/easy/turing-smart-screen-python-2.0.3/library/lcd_comm_rev_b.py", line 90, in Hello
    self.SendCommand(Command.HELLO, payload=hello, bypass_queue=True)
  File "/home/easy/turing-smart-screen-python-2.0.3/library/lcd_comm_rev_b.py", line 80, in SendCommand
    self.WriteData(byteBuffer)
  File "/home/easy/turing-smart-screen-python-2.0.3/library/lcd_comm.py", line 73, in WriteData
    self.lcd_serial.write(bytes(byteBuffer))
  File "/usr/local/lib/python3.9/dist-packages/serial/serialposix.py", line 615, in write
    raise PortNotOpenError()
serial.serialutil.PortNotOpenError: Attempting to use a port that is not open

JoBiclou, 2022-11-04:
and, now, the "main.py" return :
# python3 main.py
13:41:07 [INFO] Loading theme Terminal from res/themes/Terminal/theme.yaml
13:41:07 [DEBUG] Static COM port: /dev/ttyS0
13:41:08 [WARNING] Device not recognised (short response to HELLO)
Traceback (most recent call last):
  File "/home/easy/turing-smart-screen-python-2.0.3/main.py", line 56, in <module>
    display.initialize_display()
  File "/home/easy/turing-smart-screen-python-2.0.3/library/display.py", line 57, in initialize_display
    self.lcd.InitializeComm()
  File "/home/easy/turing-smart-screen-python-2.0.3/library/lcd_comm_rev_b.py", line 117, in InitializeComm
    self.Hello()
  File "/home/easy/turing-smart-screen-python-2.0.3/library/lcd_comm_rev_b.py", line 95, in Hello
    if response[0] != Command.HELLO or response[-1] != Command.HELLO:
IndexError: index out of range

mathoudebine, 2022-11-14:
Are you sure about the serial port? On Linux for the Turing smart screen it usually is `/dev/ttyACM0` / ACM1 / ... but not `/dev/ttyS0`
After you have checked, you can hardcode the serial port in simple-program.py l.19
https://github.com/mathoudebine/turing-smart-screen-python/blob/main/simple-program.py#L19
then run it again

guglovich, 2023-12-18:
I only have /dev/ttyS0 available too