mathoudebine/turing-smart-screen-python #81: 5" Support

MrFr1day, 2022-11-07:
Hi,

It seems that the 5" version does not use the same protocol as the 3.5", if it set it to revision A, it will connect and send the data to the monitor (I can see it in the sniffer), but nothing will show up on the monitor.

While revision B will not connect, and the following will show up:
> [WARNING] Device not recognised (short response to HELLO)
> Traceback (most recent call last):
>   File "F:\tss\main.py", line 56, in <module>
>     display.initialize_display()
>   File "F:\tss\library\display.py", line 57, in initialize_display
>     self.lcd.InitializeComm()
>   File "F:\tss\library\lcd_comm_rev_b.py", line 118, in InitializeComm
>     self.Hello()
>   File "F:\tss\library\lcd_comm_rev_b.py", line 95, in Hello
>     assert response, "Device did not return anything"
> AssertionError: Device did not return anything

A couple of things I've noticed:
- The monitor seems to have a built-in memory, because the background video is uploaded to it even though I've not added an SD card.
- The background video will continue to play/loop even if the app (UsbMonitorL) is closed.

Which makes me wonder, if the background is uploaded at the beginning and set to play, does that mean that the transparent-background for all elements on top of it (text, bars ...) are rendered by the monitor it self? if so, could this reduce the processing/data sent by the PC?

Thanks.

mathoudebine, 2022-11-07:
Thanks for investigating!
Quick question: did you change the display width/height in the `config.yaml` when testing with revision A?

If the elements are displayed over the background video by the default software it means indeed that the 5" version supports transparency in bitmaps, so the protocol is likely to be different from the 3.5" version that does not support transparency.

The next step would be to uncompile UsbMonitorL.exe but it has been deliberately obfuscated with Dotfuscator, so i'm not sure if it can be done.
Otherwise the serial link between computer and display must be analyzed to try to understand the protocol (as it has been done for the revision B).

MrFr1day, 2022-11-07:
- Yes, I tried both actually, I also had to set the COM port manually.

- How was it analyzed? anything a beginner can try?
I can log the comms while using a simple white-background theme as start if that helps, though I don't know which app to use.

mathoudebine, 2022-11-14:
The original protocol for the revision A was analyzed by uncompiling UsbMonitor.exe first release (it was not obfuscated). For the revision B, @gerph did the serial link analyze so maybe he can give you more information?

gerph, 2022-11-14:
As I remember, I used https://freeserialanalyzer.com/ and a couple of others whilst the exe was running on Windows to capture the output. Then I stepped through the output looking for things that I could recognise as being patterns - the framing of using a start and stop byte that were the same makes the commands easier to see. With that in mind, and from looking at the style of the original model A code - and assuming that the later model was developed with what was learnt from the earlier one - I could spot things like the brightness and the 'place some graphic here' commands.

The sequence was involved - load the serial port analyzer software and display software up on Windows, enable the display in the software so that something is shown and I've got a lot in the port analyzer. Stop the display software. Save off the output from the analyzer and step through the sequences of bytes until I can understand what's being sent. It took about an afternoon to get to understand the protocol and how things worked. Once I'd got an idea of the commands I could also try variants of them - what happens if I put some extra values in the packets tha are sent? what happens if I sent a packet that I've never seen but which could do something (that's how the orientation commands were found)?

Where is the 5" version available from? It'd be interested in grabbing one and trying it out.

MrFr1day, 2022-11-17:
Hi @gerph,

You can search for it in AliExpress using something like "5.0 inch IPS AIDA64", just make sure it isn't actually for AIDA64, because despite the the listing title, most of them are not for AIDA64, as they explain in the details.

Here are a few vendors I found:
https://www.aliexpress.com/item/1005003259563629.html
https://www.aliexpress.com/item/1005004495210303.html
https://www.aliexpress.com/item/1005004268341329.html