mathoudebine/turing-smart-screen-python #82: Your CPU temperature is not supported yet

Yawllen, 2022-11-10:
When starting the program выдает следующие

[image]

Win 11. Intel core i7 11700kf

Yawllen, 2022-11-10:
Так же монитор зависает при запуске любой программы

spectra9, 2022-11-12:
Although not a bug, since they clearly document it that CPU temp does not work on Windows, I do wish that they would eventually release a version that do. I mean, CPU temp is probably the first thing, or maybe second to GPU temp that people would want to monitor. A system monitor without the ability to monitor CPU temp, is just plain odd and frankly quite useless for me

mathoudebine, 2022-11-14:
Hi, for the "CPU temperature not supported yet" message, I'm currently working on a new way to get CPU sensors from LibreHardwareMonitor instead of Python3 libraries. You can follow progress here: #52

@Yawllen can you try to run `simple-program.py`? Does it work with your screen? You may need to edit the `simple-program.py` file first to change `REVISION` or `COM_PORT` if needed.

spectra9, 2022-12-09:
Downloaded and testing. After changing COM port setting (AUTO never works in my case), the temp for CPU is showing and seems to be in line with what RTSS is showing. Also I get message  [WARNING] Network interface 'eth0' not found and 'wlan0' not found messages repeatedly, probably need to play around with the network interface name. Other than that, looking good so far, will test some more. 

Great work @mathoudebine, really appreciate it you keeping this project alive

spectra9, 2022-12-09:
Found test_librehardwaremonitor.py under LibreHardwareMonitor folder. Would be very cool if we could select which sensor to read from this list :-)

spectra9, 2022-12-09:
I think I found the problem with the CPU clock calculation. It's calculating the mean values of every Core# clocks, including the effective core clocks, which skew the results. So in order to make in line with RTSS value, I change the following lines

```
  for sensor in cpu.Sensors:
      if sensor.SensorType == Hardware.SensorType.Clock and "Core #" in str(sensor.Name):
          frequencies.append(float(sensor.Value))
  return mean(frequencies)

```

to this

```
for sensor in cpu.Sensors:
    if sensor.SensorType == Hardware.SensorType.Clock and "Core #" in str(sensor.Name) and "Effective" not in str(sensor.Name):
        frequencies.append(float(sensor.Value))
return max(frequencies)

```

spectra9, 2022-12-11:
Hey Matt, thanks for the reply.
My CPU is AMD 5900X for reference and yes I do have the "Effective" core clock metric

I have updated the network interface name, so that's working fine now.

I've also tinkered around in replacing the GPU Mem Used metric to GPU Core Clock, was too lazy to add a new one so I just replace the one I don't use lol.
Next maybe onto some date/time formatting if I have time -> would be nice to have this configurable in the config too, along with some alignment choice hehe. But currently it's showing all it needed to show for my need

mathoudebine, 2022-12-12:
Thanks again for your feedback, i've opened a feature request for custom date/time: #99 
And one for the GPU Clock as a new metric #103

closed 2022-12-13 by mathoudebine