mathoudebine/turing-smart-screen-python #86: Incorrect CPU clock

Rand0m80, 2022-11-20:
**Describe the bug**  
Incorrect CPU clock is displayed

**To Reproduce**  
Normal run

**Expected behavior**  
After running project, CPU frequency displayed on my xuanfang display as 3.7 GHz and dont changing.
In the same time MSI hardware monitor correctly displayed frequency as changing value from 900 to 4795 MHz.

**Screenshots / photos of the Turing screen**  

[image]

**Environment:**  
 - Release or branch of this project [2.1.0]
 - OS with version [Windows 10 ver. 21H2]
 - Python version [Python 3.11.0]
 - Hardware [CPU Intel i5-9600K OC to 4.8 GHz, MB MSI Z370 SLI+]

**Additional context**  
Maybe problem somewhere here 
[image]

Rand0m80, 2022-12-12:
All works fine. Thank you.

closed 2022-12-12 by Rand0m80

mathoudebine, 2022-12-13:
Fixed in 2.2.0 release