mathoudebine/turing-smart-screen-python #87: Installation issue on debian (proxmox)

gtrhayrug, 2022-11-20:
Hi,
First, i want to thank you for this contribution !
I'm using Proxmox 7.2 (debian 11) uptodate.
I've gitclone the rep, done.

python3 is installed.
requirements from directory done.
i've found the usb port (using dmesg) : ttyACM0
ip a give me vmbr0.
It is a B version.

So, i have modified the config.yaml with 

`---
config:
  # Configuration values to set up basic communication
  # Set your COM port e.g. COM3 for Windows, /dev/ttyACM0 for Linux...
  # COM_PORT: "/dev/ttyACM0"
  # COM_PORT: "COM3"
  # COM_PORT: "AUTO"
  COM_PORT:"/dev/ttyACM0"

  # Theme to use (located in res/themes)
  # THEME: Terminal
  # THEME: Landscape6Grid
  # THEME: Cyberpunk
  # THEME: bash-dark-green
  THEME: 3.5inchTheme2

  # Network metrics
  # Put the interface name or let it blank if the card does not exist
  # Linux/MacOS interfaces are named "eth0", "wlan0", "wlp1s0", "enp2s0"...
  # For Windows use the interfaces pretty name: "Ethernet 2", "Wi-Fi", ...
  # ETH: "eth0" # Ethernet Card
  ETH: "vmbr0"
  # WLO: "wlan0" # Wi-Fi Card

display:
  # Display resolution in portrait orientation
  # Do not use this setting to rotate display! Display orientation is managed by themes
  DISPLAY_WIDTH: 320
  DISPLAY_HEIGHT: 480

  # Display Brightness
  # Set this as the desired %, 0 being completely dark and 100 being max brightness
  # Warning: screen can get very hot at high brightness!
  BRIGHTNESS: 20

  # Display revision: A or B (for "flagship" version, use B) or SIMU for simulated LCD (image written in screencap.png)
  # To identify your revision: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Hardware-revisions
  REVISION: B`

but when i want to start the program with python3 config.yaml:
`  File "/root/turing-smart-screen-python/config.yaml", line 1
    ---
       ^
SyntaxError: invalid syntax`

if i want to start "simple program" with python3 simple-program.py
`Selected Hardware Revision B (XuanFang screen version B / flagship)
19:03:04 [ERROR] Cannot find COM port automatically, please set it manually in config.yaml`

mathoudebine, 2022-11-20:
Hello french compatriote! 🇫🇷

In the file `config.yaml` you need to keep `WLO` entry, if you do not have a wireless LAN card you can just let the field empty like that:
`WLO: ""`

You also need to have a space between `COM_PORT:` and the port, like that:
`COM_PORT: "/dev/ttyACM0"`
the yaml format is restrictive...

the command to launch the system monitor program is:
`python3 main.py`
because the program in Python is in `main.py`
The file `config.yaml` is only the configuration file, it is loaded automatically by `main.py` at startup

You can have a look at the wiki here: https://github.com/mathoudebine/turing-smart-screen-python/wiki/t you will find all the commands in "System Monitor > How to start" it will also help you to install dependencies if needed.

For `simple-program.py`, if you want to use it you have to manually edit the file `simple-program.py` directly with a text editor to change settings at the top of the file, because simple-program does not use the `config.yaml` for configuration

Keep me informed if you have other issues!

gtrhayrug, 2022-11-21:
Hi,
Thanks !
After config.yaml (wlan) working !

First problem, my computer has 32 Go but screen reports only 16.
```

               total        used        free      shared  buff/cache   available
Mem:        32547112    25188412     1172788       66072     6185912     6809836
Swap:       33554428     2635008    30919420

```

Maybe stupid question, but  just want to finish to test the different layouts.
But i don't find to stop the python process clearly. (i've done `kill -9 xxxx `after kept xxxx with `ps -ef | grep python3`
Even if grep the process, the port USB still be blocked but the first launched process.
```

06:18:15 [INFO] Loading theme Terminal from res/themes/Terminal/theme.yaml
06:18:15 [DEBUG] Static COM port: /dev/ttyACM0
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/turing-smart-screen-python/main.py", line 19, in <module>
    import library.scheduler as scheduler
  File "/root/turing-smart-screen-python/library/scheduler.py", line 8, in <module>
    import library.stats as stats
  File "/root/turing-smart-screen-python/library/stats.py", line 22, in <module>
    from library.display import display
  File "/root/turing-smart-screen-python/library/display.py", line 97, in <module>
    display = Display()
  File "/root/turing-smart-screen-python/library/display.py", line 37, in __init__
    self.lcd = LcdCommRevA(com_port=CONFIG_DATA['config']['COM_PORT'],
  File "/root/turing-smart-screen-python/library/lcd_comm_rev_a.py", line 25, in __init__
    self.openSerial()
  File "/root/turing-smart-screen-python/library/lcd_comm.py", line 71, in openSerial
    self.lcd_serial = serial.Serial(lcd_com_port, 115200, timeout=1, rtscts=1)
  File "/usr/local/lib/python3.9/dist-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/usr/local/lib/python3.9/dist-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
```

Last question to transform your python code into a service (mode simple to start, stop, restart), what is the best way ? and to add it at the startup. Maybe adding in the cron is better (after all booting), isn't it ?

mathoudebine, 2022-12-12:
Hi @gtrhayrug 

Did you find a fix for your issue with the incorrect RAM size?

To clean stop the process, you can send a SIGTERM (15) to the program, which is the same as doing a Ctrl+C in the command window: `kill -15 xxx` or `kill -s SIGINT xxx`
This signal will be caught by the system monitor main.py to stop cleanly. This way, it will not stop the program while a frame is being sent to the screen.
This should fix your issue.

To run the program as a service, you can check this wiki page that has been updated: 
https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-%3A-how-to-start#6-run-system-monitor-at-startup

gtrhayrug, 2022-12-14:
OK, i've already used the principle of kill -9 or -15.
I have tried to use service but even if classical method works (python3 main.py) if i've tried, always the message :

```
     Loaded: loaded (/etc/systemd/system/turing-smart-screen-python.service; enabled; vendor preset>
     Active: failed (Result: exit-code) since Wed 2022-12-14 09:45:11 CET; 2s ago
    Process: 3540408 ExecStart=python3 main.py (code=exited, status=200/CHDIR)
   Main PID: 3540408 (code=exited, status=200/CHDIR)
        CPU: 4ms

Dec 14 09:45:11 SRV-DOMO systemd[1]: turing-smart-screen-python.service: Scheduled restart job, res>
Dec 14 09:45:11 SRV-DOMO systemd[1]: Stopped Turing Smart Screen.
Dec 14 09:45:11 SRV-DOMO systemd[1]: turing-smart-screen-python.service: Start request repeated too>
Dec 14 09:45:11 SRV-DOMO systemd[1]: turing-smart-screen-python.service: Failed with result 'exit-c>
Dec 14 09:45:11 SRV-DOMO systemd[1]: Failed to start Turing Smart Screen.
```

mathoudebine, 2023-05-24:
Closing this issue as Exit conditions has been improved in latest versions. Tag me or open another bug if you still have specific issues

closed 2023-05-24 by mathoudebine