mathoudebine/turing-smart-screen-python #106: Cant for the life of me to get it to run on boot!

Neuropsihopuppu, 2022-12-13:
Any help would be apreciated i am lost here

**Describe the bug**  
Can't get to run on boot!

**To Reproduce**  
Steps to reproduce the behavior:
Followed the exact instructions:

Windows
Because the system monitor needs to be run as administrator on Windows, the only way to start it at boot is by creating a Scheduled Task.

From your search bar, search and open "Task Scheduler"
From the scheduler, select "Create Basic Task" from the right panel.
From the trigger page, select "When I log on".
From the action page, select "Start a program".
In the "program/script" field, browse and select the pythonw.exe program (note the 'w' in the name). It is usually located in C:\Users\YourUser\AppData\Local\Programs\Python\Python3XX\.
In the "Add arguments" field, enter main.py
In the "Start in" field, enter the complete path to the folder of this project e.g. C:\Users\You\Documents\turing-smart-screen-python\
In the "Finish" tab, select "Open the properties dialog for this task..."
In the Task Properties dialog on the General tab, select "Run with highest privileges"

**Expected behavior**  
Program Runs on start up

**Environment:**  
 - Revision of this project :2.2.0
 - OS with version :Windows 11
 - Python version :python 3.9
 - Hardware [e.g. Intel CPU, Nvidia GPU, Raspberry Pi 3 B+ ...]

1080papaya, 2022-12-14:
@Neuropsihopuppu I'm not associated with the team for this but I think you can get around this with having task scheduler run the script through CMD or a batch file that loads the script .

E.g. a batch file that will look something like py main.py

Neuropsihopuppu, 2022-12-18:
> 

that does not work because python doesent run as admin if started from the batch file.

sorry for my bad english

RobinPerson, 2022-12-19:
If you installed Python from the Microsoft Store, uninstall it and download and install Python from the official website. I had this issue as well. With that installed, I can add and run the script with the following settings:
Command: python
Parameter: .\main.exe
Start in: [location of the script]

Now I still have the issue that trying to start it hidden/in the background on Windows is impossible as of now. It's either https://github.com/mathoudebine/turing-smart-screen-python/issues/23 with the Python libraries, which causes a ton of performance issues, or the script not running at all with OpenHardwareMonitor's libraries, because that won't start without a window present (I've tried several methods, including pythonw).

mathoudebine, 2022-12-19:
With OpenHardwareMonitor on Windows, the projet may not load for security reasons (cannot load dll from untrusted sources by default). There is a workaround in the wiki : https://github.com/mathoudebine/turing-smart-screen-python/wiki/Troubleshooting#windows-cannot-load-librehardwaremonitordll-loadfromremotesources

Does the program run when started manually with `pythonw.exe`?

mathoudebine, 2022-12-19:
I also added a feature request to add a tray icon, to be sure the program is running in the background or not: #111

RobinPerson, 2022-12-19:
I forgot to mention that indeed when testing out LHM, you have to make sure the setting for running it with highest privileges is turned on as well. But that does not fix the issue for me.

As for running it with pythonw.exe outside of Task Scheduler, I have not had any success with it using LHM. It returns out of the program without ever initializing the display. When using the Python libraries it does work with pythonw.exe but that results in the window opening and closing every second again.

Atgp540, 2023-02-01:
I have found some work around for LHM service but I also still cannot get it to run at start up.

I maid a main.pyw so it doesn't show terminal. 
I enabled run as admin on lhm app ( right click properties run as admin or advanced admin for all users )
I then used the pyinstaller extension to convert the py.w to a .exe which I have the same run as admin treatment.

This .exe will now start the program create the Taskbar icon and work as expected 
how ever
It will not run if placed in shell:startup and won't run if created as a schedule either. 
I also tried to make a .bat to run it but after trying to add it as a service it gave the above failure also

But a good thing is you can short xut the .exe to your desktop and run it with out right click as admin and edit the config files exit and restart the exe and changes will show up.

mathoudebine, 2023-05-24:
Closing this bug as inactive since February, tag me or open another bug report if you still have issues

closed 2023-05-24 by mathoudebine