mathoudebine/turing-smart-screen-python #107: New Board Revision?

airfield20, 2022-12-16:
**Describe the bug**  
My board is not shown in the images of the board revision, however Rev A configuration works on linux (when reset is disabled)

**To Reproduce**  
Steps to reproduce the behavior:
N/A

**Expected behavior**  
A clear and concise description of what you expected to happen.
N/A

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the Turing screen to help explain your problem.
[image]
 

**Environment:**  
 - Ubuntu 20

**Additional context**  
board was purchased from amazon, shipped from china.

mathoudebine, 2022-12-16:
I've never seen this board! Must be a new revision A. Can you send the link where you bought it please?
I will update the documentation with your photo if that's okay

airfield20, 2022-12-16:
hokistudio 3.5 Inch 480x320 IPS TYPEC Screen for Aida64 PC CPU Monitor Computer Temperature with Metal Shell https://a.co/d/4wBgbpV

Sure that's fine with me

mathoudebine, 2022-12-22:
Documentation updated

closed 2022-12-22 by mathoudebine