mathoudebine/turing-smart-screen-python #116: Create an installer for Windows

mathoudebine, 2022-12-22:
**Is your feature request related to a problem? If so, please describe the problem.**  
Installing this project and running it at boot must be done manually

**Describe the feature / solution to your problem you'd like**  
An all-in-one installer for Windows

**Describe alternatives you've considered / and or tested**  
Using the documentation https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-%3A-how-to-start but it's not easy for everybody

**Environment:**  
 - Windows

Cyruz143, 2022-12-22:
Would be a nice idea to have an update mechanism built in if you're going to package this for Windows.

mathoudebine, 2023-10-20:
Hi @MarkOliver1959 I look forward to it! This will be a nice addition to this project, and could simplify the installation procedure for people less comfortable with technology

closed 2025-01-29 by mathoudebine