mathoudebine/turing-smart-screen-python #119: seperate `lcd_comm` as library/driver

anwar3606, 2022-12-29:
**Is your feature request related to a problem? If so, please describe the problem.**  
One of my projects just wants the `library/lcd/` files. I don't need any other files/folders like res, tools.. etc.
Those `lcd_comm` classes are used as drivers for this device. 

**Describe the feature/solution to your problem you'd like**  
I want to use those lcd_comm classes as pip packages

**Describe alternatives you've considered / and or tested**  
Currently, I had to copy those files into my project. Which will be harder to maintain in the future. 

Is there any better way to do it correctly? 
If want do a pull request, how should I proceed?

mathoudebine, 2023-01-05:
Good idea to separate the library and the system monitor program!
A new repository has to be created
I have to think about a valid name for the package, like `pyturing` or `turing-lcd`

rikkuness, 2024-04-21:
This would also be really useful for issue tracking etc. too! I only use the library to use the screens with my own code and when looking to see if issues have been reported already the majority of issues are around theming and monitoring things. Also thanks for the great library!