mathoudebine/turing-smart-screen-python #138: LED backplate color control only works on one

Rollbacke, 2023-01-08:
**Describe the bug** 
On the XuanFang (rev B), when you configure a color for the blackplate LEDs, it only configures one (not all).
It's not a problem of my XuanFang : on their software, all the LEDs are OK.
I haven't had time to look at the code yet.

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Revision : main
 - OS with version : tested on Windows (11) and Arch Linux (Manjaro)
 - Python 3.10

Rollbacke, 2023-01-10:
Tonight, I don't have the problem.
I don't understand
If I can identify the problem, I'll reopen the issue.

closed 2023-01-10 by Rollbacke

mathoudebine, 2023-01-12:
Sometimes i have this issue with my screen also. I did not find the cause

Rollbacke, 2023-01-13:
I haven't had the problem since.
I wonder if it's permanent, for example, is it just a miss, and in that case, we would just have to send the instruction twice.

mathoudebine, 2023-01-13:
If you have the issue again, you can try to test to send instruction twice to see if it fix the issue

Atgp540, 2023-01-31:
can you remove the back panel and take a photo of the smd rgb leds? I'm curious which ones to buy to for my B variant with out leds installed.

also cannot seem to find a photo of a LED variant online.

mathoudebine, 2023-01-31:
I'm note sure if the backplate can be removed, i think it's glued on the back. If you're searching for the variant with LEDs it's usually called "flagship version"

Atgp540, 2023-01-31:
My xfaung has 2 Phillips head screws holding the back panel on with a light
defused floating ( not glued ) maybe that one above is different?

inside mine looks like the B variant in the what display do I have section
which has spots for led placement they even put some soilder but not the
smd led.

Atgp540, 2023-01-31:
Here is a shot of mine
I have notice a variance in mine to the one photod in the what version section.

I am missing a resistor on  r3 I wonder what that's for? Probably led control. I'm also missing the leds which I have determined are smd 0805 rgb led from measuring.

Would still be nice to see a shot of a working led flag ship model for science. 
[image]
[image]

Rollbacke, 2023-02-01:
Hey,
I have a photo of my old version B (without LED) that I had found on Aliexpress (it no longer works, the screen lights up but does not display anything).
Unlike you, I don't have a resistor on the R6 but I have one on the R3. It's weird.

mathoudebine, 2023-02-01:
Same for me on my revision B, i also have the R3 resistor. It's the same as R1.
As for the flagship revision with LEDs, here is a photo of one of the SMD LED unfortunately not much to see
[image]

Atgp540, 2023-02-01:
This photo is exactly what I needed to verify the smd led 0805 rgb smd led is the part.

I would assume the resistor is probably for led control buy I'm thinking of retrofitting some in to find out... this could be a while tho

Thanks for the photos friends