mathoudebine/turing-smart-screen-python #139: The GPU used memory can be different depend on the hardware

Rollbacke, 2023-01-08:
**Describe the bug**  
On my computer, the metric for the GPU used memory is named `D3D Dedicated Memory Used`.

```
Fullscreen FPS : -1.0
D3D Dedicated Memory Used : 1164.87109375
D3D Shared Memory Used : 196.86328125
D3D 3D : 3.724090337753296
D3D High Priority 3D : 0.0
D3D High Priority Compute : 0.0
D3D True Audio 0 : 0.0
D3D True Audio 1 : 0.0
D3D Compute 3 : 0.0
D3D Compute 0 : 0.0
D3D Compute 1 : 0.0
D3D Video Decode 1 : 0.0
D3D Video Encode 0 : 0.0
D3D Video Codec 0 : 8.5394926071167
D3D Copy : 0.0
D3D Copy : 0.0
D3D Timer 0 : 0.0
D3D Security 1 : 0.0
GPU Package : 38.0
GPU Core : 57.0
GPU Memory : 64.0
GPU VR VDDC : 49.0
GPU Hot Spot : 58.0
GPU Core : 168.0
GPU SoC : 689.0
GPU Memory : 1744.0
GPU Fan : 843.0
GPU Fan : 24.0
GPU Core : 0.7250000238418579
GPU SoC : 0.7930000424385071
GPU Memory : 0.8500000238418579
GPU Core : 7.0
GPU Memory : 2.0
```

**Environment:**  
 - Revision of this project: main
 - Windows 11
 - Python 3.10
 - Hardware: AMD Radeon 5700 XT

**Additional context**  
I don't know if it's specific to my GPU card...
What bothers me is that I don't see a metric about the total memory.
It's a problem to calculate the % used...

mathoudebine, 2023-01-13:
Hello @Rollbacke
There is indeed the `D3D Dedicated Memory Used` sensor for all graphic cards.
It is not quite the same as `GPU Memory Used` : 
> Direct3D memory used: This is the amount of memory used by the GPU via DirectX/Direct3D. It doesn't cover memory usage via other interfaces (like OpenCL for example).
Unfortunately, AMD doesn't seem to offer an universal way how to report exact memory usage of the GPU (like nVidia), so this is currently the only way how to get a picture of memory usage.
>
>"GPU Utilization" is read straight from the GPU internal registers, so this should be the more precise value and it should cover the entire GPU.
"D3D Usage" is read via Direct3D, so it covers only this interface.
https://www.hwinfo.com/forum/threads/what-is-d3d-usage.759/

So the D3D Dedicated Memory Used will always be lower that the total GPU memory used, it can be seen on Nvidia cards that have both sensors:
[image]

I agree with you, it can still be used for AMD GPU, better than nothing 😉 

For AMD GPU there is also no sensors to get the total memory amount, in LHM or with Python libraries. I think it is a limitation from the manufacturer. This means the GPU Memory cannot be displayed as a graph for these platforms.
I will think about adding a manual entry for GPU Memory in `config.yaml`

Rollbacke, 2023-01-13:
Hi @mathoudebine,

> Direct3D memory used: This is the amount of memory used by the GPU via DirectX/Direct3D. It doesn't cover memory usage via other interfaces (like OpenCL for example).
> Unfortunately, AMD doesn't seem [...]

Oh my god 🤦‍♂️...

> So the D3D Dedicated Memory Used will always be lower that the total GPU memory used, it can be seen on Nvidia cards that have both sensors: [image]
> 
> I agree with you, it can still be used for AMD GPU, better than nothing wink

Yeap. We can add a condition for AMD GPU which use  "D3D Dedicated Memory Used".

> For AMD GPU there is also no sensors to get the total memory amount, in LHM or with Python libraries. I think it is a limitation from the manufacturer. This means the GPU Memory cannot be displayed [...]

I was thinking about it... I didn't want to suggest it, but hey, I don't think there's too much of a solution...

closed 2023-02-06 by mathoudebine