mathoudebine/turing-smart-screen-python #144: error with requirements.txt

Huyrrrr, 2023-01-17:
I followed "System monitor : how to start" instructions, but when I install the requirements.txt, i have that issue:
[image]

Huyrrrr, 2023-01-18:
thanks for your reply. My W10 is a 64bit version. 
I tried with verson number removed but still the same
Here is the result of `python3 -m pip -vvv install pywin32` 
[image]

mathoudebine, 2023-01-18:
I think i see the issue in your last capture: you are using the python 3.10 intepreter that comes from Mingw, not a native Python interpreter for Windows.

If this is not expected, you will need to start the Python native intepreter that you installed previously, it should be on `C:\Users\your_username\AppData\Local\Programs\Python\Python3XX\python.exe`

If you still want to use the Python from Mingw, i did not test it myself so it may work or not. You will need to install this Mingw package first https://packages.msys2.org/base/mingw-w64-python-pywin32 then you can restart the installation command. It will detect that pywin32 is already installed and will skip it.

Huyrrrr, 2023-01-18:
Thanks it works !

mathoudebine, 2023-01-18:
Nice! Closing the issue but feel free to comment or open a new one if you have other problems 👍

closed 2023-01-18 by mathoudebine