mathoudebine/turing-smart-screen-python #148: The ability to change the time it takes to update the status on the screen

ftk789, 2023-01-24:
**Is your feature request related to a problem? If so, please describe the problem.**  
The feature is not a problem and it's just an addition.

**Describe the feature / solution to your problem you'd like**  
The feature is the ability to change the time to update the status on the screen

**Describe alternatives you've considered / and or tested**  
Haven't tested anything, But since for an example the CPU Load takes 1 second each update, And for the rest of course, Is it possibly to have the ability to decrease/increase that time?
**Screenshots / photos & mockups of the Turing screen**  
No need since it's self explanatory 

**Environment:**  
Windows 10.
Python leatest

**Additional context**  
Thank you for an amazing project.

mathoudebine, 2023-01-25:
Hello!
For each sensor displayed you can change the time between each refresh of the value:
Go to the folder of your selected theme in `res/themes/` and open the `theme.yaml` file
Search for the sensor you want to change e.g. for the CPU frequency (Ghz) it is in `STATS > CPU > FREQUENCY`:
```yaml
STATS:
  CPU:
    [...]
    FREQUENCY:
      # In seconds. Longer intervals cause this to refresh more slowly.
      # Setting to lower values will display near real time data,
      # but may cause significant CPU usage or the display not to update properly
      INTERVAL: 5
      TEXT:
      [...]
```
You can see here an `INTERVAL` value that is a delay in seconds between two refresh of the sensor value on screen.
Here by default the CPU frequency is refreshed every 5 seconds on the display.
You can change this interval for every sensor you want on this file!

Be advised though that refreshing a sensor more frequently also means increased CPU usage to get the data, compute the new picture and display it on the screen. So i advise you no to change all sensors to 1sec delay. You can make some tests and find the best compromise 🙂

ftk789, 2023-01-25:
Thank you very very much for this info! It works like a charm, Thank you for an amazing program!

closed 2023-01-25 by ftk789