mathoudebine/turing-smart-screen-python #150: Cant start main.py from Terminal or Task Scheduler

johndomenico, 2023-01-26:
When I try to start the program via the terminal I get the following error message:

File "C:\Users\XX\Downloads\tur\main.py", line 48, in <module>
    import library.scheduler as scheduler
  File "C:\Users\XXX\Downloads\tur\library\scheduler.py", line 27, in <module>
    import library.config as config
  File "C:\Users\XXX\Downloads\tur\library\config.py", line 37, in <module>
    CONFIG_DATA = load_yaml("config.yaml")
                  ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\XXX\Downloads\tur\library\config.py", line 31, in load_yaml
    with open(configfile, "r") as stream:
         ^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'config.yaml'

Its just working when its started from VS Code.

 I checked out the newest Version, and iam on Windows11

mathoudebine, 2023-02-06:
Hello!
What is the command you use to start the program from the terminal?

mem1991, 2023-03-02:
i seem to also be having the same issue when trying run from startup i have tested the config and followed instructions and i can confirm that the inital setup works and works flawlessly when running from cmd prompt however when trying to run task scheduler to initialise at login it simply doesn't open display completely turned off this is with the B revision of display,

this is following exactly from 7. Run system monitor at startup but only changing the directory?

mathoudebine, 2023-05-11:
Hi, did you follow these 2 steps when creating your Scheduled Task?

> * In the "Add arguments" field, enter main.py
> * In the "Start in" field, enter the complete path to the folder of this project e.g. C:\Users\You\Documents\turing-smart-screen-python\

A common error is to add the path into the "Add arguments" field and left the "Start in" field blank, which can show this error

mathoudebine, 2023-05-24:
Closing this issue for inactivity, tag me or open a new issue if you still have problems

closed 2023-05-24 by mathoudebine