mathoudebine/turing-smart-screen-python #155: Tray icon isn't displayed when called from pythonw

Cyruz143, 2023-02-06:
**Describe the bug**  
Tray icon isn't displayed when called from pythonw. Potentially a limitation of pythonw, works fine if you call from command line with python.

**To Reproduce**  
Steps to reproduce the behavior:
1. Load via pythonw, no tray icon

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the Turing screen to help explain your problem.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Revision of this project [e.g. 1.1.1, `main` branch, specific commit]
 - Windows 10 x64
 - Python 3.10.6
 - AMD CPU + Nvidia GPU

**Additional context**  
N/A

mathoudebine, 2023-02-07:
How do you start pythonw?
When I use this command line `pythonw.exe main.py` the tray icon is displayed on my computer with the same setup as yours.
Are you sure the icon is not in the hidden icons panel?

Cyruz143, 2023-02-07:
Crap, should have checked that. It's fine direct from PS but if you use the task scheduler to start it, it doesn't fire the tray icon.

Cyruz143, 2023-02-07:
Ok, closing this. Removed the task and recreated and now it works fine. Can't quite explain why!

closed 2023-02-07 by Cyruz143