mathoudebine/turing-smart-screen-python #161: SENSOR error

Ultra769, 2023-02-11:
I place it onto anything i can't, if admin I set access then sensor signals a mistake

 # Hardware sensors reading
  # Choose the appropriate method for reading your hardware sensors:
  # - PYTHON         use Python libraries (psutils, GPUtil...) to read hardware sensors (supports all OS but not all HW)
  # - LHM            use LibreHardwareMonitor library to read hardware sensors (Windows only - NEEDS ADMIN RIGHTS)
  # - STUB / STATIC  use random/static data instead of real hardware sensors
  # - AUTO           use the best method based on your OS: Windows OS will use LHM, other OS will use Python libraries
  HW_SENSORS: AUTO
----------------------------

C:\Users\ultra\Downloads\turing-smart-screen-python-main\a>python main.py
13:28:13 [INFO] Loading theme 3.5inchTheme2 from res/themes/3.5inchTheme2/theme.yaml
13:28:13 [DEBUG] Auto detected COM port: COM5
13:28:13 [DEBUG] Found LibreHardwareMonitorLib 0.9.1.0
13:28:13 [ERROR] Program is not running as administrator. Please run with admin rights or choose another HW_SENSORS option in config.yaml

----------------------------------------------------------------------------------------------------------------------------------------------

 # Hardware sensors reading
  # Choose the appropriate method for reading your hardware sensors:
  # - PYTHON         use Python libraries (psutils, GPUtil...) to read hardware sensors (supports all OS but not all HW)
  # - LHM            use LibreHardwareMonitor library to read hardware sensors (Windows only - NEEDS ADMIN RIGHTS)
  # - STUB / STATIC  use random/static data instead of real hardware sensors
  # - AUTO           use the best method based on your OS: Windows OS will use LHM, other OS will use Python libraries
  HW_SENSORS: ADMIN

C:\Users\ultra\Downloads\turing-smart-screen-python-main\a>python main.py
13:29:24 [INFO] Loading theme 3.5inchTheme2 from res/themes/3.5inchTheme2/theme.yaml
13:29:24 [DEBUG] Auto detected COM port: COM5
13:29:24 [ERROR] Unsupported SENSORS value in config.yaml

mathoudebine, 2023-02-11:
Hello, ADMIN is not a valid value for HW_SENSORS property. The possible values are stated in the comment above the property.

Basically on Windows you have 2 choices:
- AUTO but you have to run the program as administrator on Windows
- PYTHON if you don't want to run the program as administrator, but the some hardware can be unsupported with this method

Ultra769, 2023-02-11:
Thanks for the quick help.
I'm very happy.
The windows command window must be run as an administrator.
 The amount of memory is not displayed.
The brightness of the screen pulsates a bit.
I deal with it.
Thanks.
The answer is very correct.

Ultra769, 2023-02-11:
All ok.
ZeroDivisionError: division by zero
THX

closed 2023-02-11 by Ultra769