mathoudebine/turing-smart-screen-python #165: Vanguard block inpoutx64.dll

kiam001, 2023-02-17:
[image]

mathoudebine, 2023-05-24:
Hi @kiam001
The file inpoutx64.sys is an old driver that is vulnerable and can be used insecurely, that's why it is blacklisted by Vanguard.
It can be installed on your computer by some software like Asus OCT, Gigabyte, OpenRGB, Logitech GHub... it is used for fan speeds or RGB leds on some hardware.
Once it has been installed on your system, LibreHardwareMonitor library will try to use it to access some hardware sensors, and it will get blocked by Vanguard.

I do not recommend to delete the file `inpoutx64.sys` because it may still be used by some software, you could have blue screen.
What you can try is deleting these 2 registry keys (please make a backup first):
- Computer\HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\inpoutx64
- Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\inpoutx64

Then reboot the computer and try again.

kiam001, 2023-05-24:
Thx i will try whan i Home. 

If the issued fixed with thats i Close the Ticket

kiam001, 2023-05-24:
Whan it works, after Test add this way To redme / Docs

mathoudebine, 2024-01-29:
Thanks, I've added this to the Troubleshooting page on the wiki: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Troubleshooting#windows-riot-vanguard-blocks-the-program-inpoutx64sys

closed 2024-01-29 by mathoudebine