mathoudebine/turing-smart-screen-python #170: The latest release breaks a custom theme (DragonBall)

mathoudebine, 2023-02-20:
The latest release breaks the custom themes that were extracted for some reason. I might be doing something stupid but the  DragonBall theme that I shared before is now not working due to this error:
```
10:42:45 [INFO] HW revision A does not support backplate LED color setting
10:42:45 [DEBUG] Drawing Image: BACKGROUND
10:42:45 [DEBUG] Drawing Text: Network
10:42:45 [DEBUG] Drawing Text: UpArrow
10:42:45 [DEBUG] Drawing Text: DownArrow
10:42:45 [DEBUG] Drawing Text: Blank
Traceback (most recent call last):
  File "D:\turing-smart-screen-python\main.py", line 161, in <module>
    display.display_static_text()
  File "D:\turing-smart-screen-python\library\display.py", line 113, in display_static_text
    self.lcd.DisplayText(
  File "D:\turing-smart-screen-python\library\lcd\lcd_comm.py", line 223, in DisplayText
    self.DisplayPILImage(text_image, x, y)
  File "D:\turing-smart-screen-python\library\lcd\lcd_comm_rev_a.py", line 157, in DisplayPILImage
    assert image_height > 0, 'Image width must be > 0'
           ^^^^^^^^^^^^^^^^
AssertionError: Image width must be > 0
```

In Theme Editor:

```
D:\turing-smart-screen-python>py theme-editor.py DragonBall
Traceback (most recent call last):
  File "D:\turing-smart-screen-python\theme-editor.py", line 107, in <module>
    refresh_theme()
  File "D:\turing-smart-screen-python\theme-editor.py", line 79, in refresh_theme
    display.display_static_text()
  File "D:\turing-smart-screen-python\library\display.py", line 113, in display_static_text
    self.lcd.DisplayText(
  File "D:\turing-smart-screen-python\library\lcd\lcd_comm.py", line 223, in DisplayText
    self.DisplayPILImage(text_image, x, y)
  File "D:\turing-smart-screen-python\library\lcd\lcd_simulated.py", line 126, in DisplayPILImage
    assert image_height > 0, 'Image width must be > 0'
           ^^^^^^^^^^^^^^^^
AssertionError: Image width must be > 0
```

_Originally posted by @YasinAl in https://github.com/mathoudebine/turing-smart-screen-python/discussions/147#discussioncomment-5050832_

mathoudebine, 2023-02-20:
@YasinAl
I'm sorry but I cannot reproduce it using your theme file from your previous post. Did you update the theme afterwards? If so, can you share your new `theme.yaml`?

YasinAl, 2023-02-20:
> @YasinAl I'm sorry but I cannot reproduce it using your theme file from your previous post. Did you update the theme afterwards? If so, can you share your new `theme.yaml`?

Here is the updated Yaml file (had to zip it to share):
[theme.zip](https://github.com/mathoudebine/turing-smart-screen-python/files/10783169/theme.zip)

If I comment out the asserts shown in the error, it works perfectly fine!

mathoudebine, 2023-02-20:
Still can't reproduce your error with your new theme, very strange...
Is it possible you also made changes in the Python code of the system monitor program?

YasinAl, 2023-02-20:
> Still can't reproduce your error with your new theme, very strange... Is it possible you also made changes in the Python code of the system monitor program?

Very weird indeed. The only thing I changed in code was commenting out the asserts to make it run. 

One other thing that seems weird is that the CPU usage is staying around 1 to 4% no matter what.

cobacdavid, 2023-02-20:
Everything is fine for me. Cannot reproduce the error (with linux).
[image]

YasinAl, 2023-02-22:
Latest commit no longer has the same issue. So whatever changed there has fixed it, but looking at the comments, nothing related was changed 😅

closed 2023-04-24 by mathoudebine