mathoudebine/turing-smart-screen-python #181: Turn the screen off before computer sleeps and back on after resume

kimholviala, 2023-02-27:
I put my computer to sleep when I'm not using it but my turing screen stays on which looks slightly weird. The USB port is powered even when the system is sleeping and since there is no software running doing screen updates it will just show the last stats until I resume my computer.

I temporarily fixed this by catching the WM_POWERBROADCAST event that gets sent to the invisible window and just shutting down the application cleanly before the computer sleeps. This makes the screen dark, and then I simply restart the app after resume using Task Scheduler. This works nicely but is a hack and it wouldn't be too hard to listen to sleep/resume power events and turning the screen off/on.

This is on Windows 10 and the latest released turing-smart-screen-python.

BTW thanks for the app, none of the offical apps worked on my flagship model B :D (although I was not planning on using them anyway, just tested quickly before getting this one)

mathoudebine, 2023-03-01:
Hi! Thanks for the idea, it would be a great addition to this program indeed!

mathoudebine, 2023-03-03:
This feature is now implemented in the main branch and will be included in the next release!

closed 2023-03-03 by mathoudebine