mathoudebine/turing-smart-screen-python #182: Add memory/SSD temperatures

kimholviala, 2023-02-27:
Would be nice to be able to show memory and SSD temperatures as those two components can run quite warm in smaller PC cases. I'm pretty sure LibreHardwareMonitor already supports those two sensors.

mathoudebine, 2023-03-01:
You're right, LHM does have sensors for disk temperatures, but the System Monitor does not really monitor disks for now, it monitors disk partitions (free/used space are partition metrics)
So before implementing this, it would be needed to add proper disk support in the system monitor, as well as keeping existing partitions support, and find a prioper way to select disk(s) & partition(s) to monitor

BustedTaint, 2023-04-27:
I have this working with LHM using the following in the LHM Sensor

    @staticmethod
    def temperature() -> int:
        disk = get_hw_and_update(Hardware.HardwareType.Storage)
        for sensor in disk.Sensors:
            if sensor.SensorType == Hardware.SensorType.Temperature and str(sensor.Name).startswith("Temperature 1"):
                return int(sensor.Value)
        return temperature