mathoudebine/turing-smart-screen-python #192: Is it possible to view fan speed  ?

kang2k10, 2023-03-27:
Is it possible to view fan speed  ?

mathoudebine, 2023-04-27:
This would be a great addition to this program! It seems supported by LibreHardwareMonitor on Windows (mostly for desktop, not laptops), for Linux and MacOS i have to find a Python library for it

smandon, 2023-05-14:
For linux, you can try bindings for lm-sensors: https://pypi.org/project/PySensors/

alexwbaule, 2023-06-06:
Using the psutils that is in sensors_python.py its possible to use:
sensors = psutil.sensors_fans()

In my case, all labels is empy, but the speeds is there:

```
{
    "nct6798": [
        [
            "",
            1106
        ],
        [
            "",
            936
        ],
        [
            "",
            3033
        ],
        [
            "",
            0
        ],
        [
            "",
            1075
        ],
        [
            "",
            1347
        ],
        [
            "",
            0
        ]
    ],
    "amdgpu": [
        [
            "",
            0
        ]
    ]
}
```
I dont know why is empty, but i will find why... 

looking in psutils source, its get the values from rmon ...

alexwbaule, 2023-06-06:
https://github.com/giampaolo/psutil/pull/2263