mathoudebine/turing-smart-screen-python #198: AMD 7900XTX not working properly

StanimirS, 2023-04-03:
**Describe the bug**  
I am getting no gpu load and gpu temperature stats, and my gpu memory is not displayed correctly. I am using amd 7900xtx. The console long for warning is :
```
4/3/2023 3:30:40 [WARNING] Your GPU load is not supported yet
4/3/2023 3:30:40 [WARNING] Your GPU temperature is not supported yet
```

**Environment:**  
master branch
-Release 2.4.0
-OS windows 11
[image]

-Python version 3.8
-AMD 7900XTX with 7950X3D and Asus rog strix x670e-f gaming wifi mobo

ooemir23, 2023-04-24:
Have a solitions ?

mathoudebine, 2023-04-24:
Hi, for new hardware support on Windows you need to get the latest Release 2.4.1 with upgraded LibraHardwareMonitor library.
This project relies on [LibreHardwareMonitor](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor) to get hardware sensors.
You also need to run the program as Administrator, otherwise it will use Python libraries to get hardware sensors which does not support recent hardware.

StanimirS, 2023-04-26:
That was not the issue. In fact I found a few issues and fixed them for my system specifically. 

So first of all my cpu is 7950X3D and it has an integrated graphics card. As a result of that I had 2 gpus. And the app always picked the integrated graphics card which libra hadrware monitor did not support temp and utilization (which is normal) so what I did was filter out the integrated gpu by its name in the get_hw_and_update in sensors_librehardwaremonitor.py.

Second issue I found was as following: I have 64GB of ram, which is more than enough to run my system without a page file. But you seem to take in virtual memory measurements as if they are the physical memory values, which actually results in decision by 0 in the Memory class. So what I did was just use the actual memory instead. You should probably add a check there and calculate based on actual memory if virtual memory is 0.

StanimirS, 2023-04-26:
I have basically 0 experience with python so all my fixes are basically hacks, but I would appreciate it if you fixed them properly.

mathoudebine, 2023-04-27:
Thanks for the feedback, I will fix the bug for the discrete GPU. I had not taken it into account when this program was created.
Can you share here your fix for the memory? Thanks

closed 2023-05-12 by mathoudebine