mathoudebine/turing-smart-screen-python #203: Time changes to 12hr format when opened with Task Scheduler

shaoqh, 2023-04-24:
**Describe the bug**  
Time displayed on screen changes to 12hr format (with AM/PM) when opened with Task Scheduler.

**To Reproduce**  
Steps to reproduce the behavior:
Starting main.py with the basic task created by turning on the PC
OR
Going into Task Scheduler and running the task associated with main.py

**Expected behavior**  
Time displays in 24hr format.

**Environment:**  
 - Release 2.4.0
 - Windows 11
 - Python 3.10

**Additional context**  
Opening the main.py from the folder containing the file manually has the time correctly in 24hr format.

mathoudebine, 2023-05-02:
Hello @shaoqh 
Do you have the correct time format if you run the program manually from a command line?
Can you also run the following Python code from a command line?
```python
import locale
print(locale.getdefaultlocale()[0])
```

shaoqh, 2023-05-11:
Hi @mathoudebine, I found out what the issue was. I was running the program under SYSTEM, not Administrators in the Task Scheduler. The time is displaying correctly now. Thank you.

closed 2023-05-11 by shaoqh

mathoudebine, 2023-05-11:
Thanks for telling us, have fun with your display!