mathoudebine/turing-smart-screen-python #220: Not recognising GPU on AMD system

theory18, 2023-05-10:
First time using your app, love it sofar however it does not recognise my RTX 4090 GPU, i believe this might be the same issue as a guy who previously had a similar issue where the app perhaps picks the integrated AMD gpu.

**4/3/2023 3:30:40 [WARNING] Your GPU load is not supported yet
4/3/2023 3:30:40 [WARNING] Your GPU temperature is not supported yet**

I looked at his solution and in the file he listed 'sensors_librehardwaremonitor' but i have no idea what to change.

My setup is 7800x3d with RTX 4090 Founders Edition.

Libre lists my GPU hardware as:
+- AMD Radeon(TM) Graphics (/gpu-amd/1)
+- NVIDIA GeForce RTX 4090 (/gpu-nvidia/0)

**Environment:**  
 - Windows 11

mathoudebine, 2023-05-11:
Hello, can you post the complete output since the program is started?
Are you running the program as Administrator?

mathoudebine, 2023-05-12:
Thanks, can you also run `external/LibreHardwareMonitor/test_librehardwaremonitor.py` with Python as administrator and post the results here? This program asks LHM library for a list of hardware and sensors on your computer.
I think your issue here is that your discrete GPU integrated in your CPU (AMD Radeon(TM) Graphics) is used for stats instead of the Nvidia GPU, it might be related to this bug #198

mathoudebine, 2023-05-12:
If you want, you can try the branch `fix/get-best-gpu-for-stats` to see if it fixes your issue. It will be merged soon

mathoudebine, 2023-05-12:
You can try the latest 2.4.2 release, it should fix your issue

closed 2023-05-12 by mathoudebine

AwekM, 2023-05-12:
Hi, it's me again. I have the same error with the program not detecting my GPU.

My specs: 
- Ryzen 5 3600
- Asrock 5700 xt
- Running Fedora Linux

This is the output using the latest release 2.4.2

acapcap@acapcap turing-smart-screen-python-main]$ python3 main.py 
Jumaat 12 Mei 2023 10:26:58  +08 [INFO] Loading theme Cyberpunk-net from res/themes/Cyberpunk-net/theme.yaml
Jumaat 12 Mei 2023 10:26:59  +08 [DEBUG] Auto detected COM port: /dev/ttyACM0
Jumaat 12 Mei 2023 10:26:59  +08 [DEBUG] Using Python 3.11.3 (main, Apr  5 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)]
Jumaat 12 Mei 2023 10:26:59  +08 [INFO] Tray icon has been displayed
Jumaat 12 Mei 2023 10:26:59  +08 [INFO] Display reset (COM port may change)...
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Auto detected COM port: /dev/ttyACM0
Jumaat 12 Mei 2023 10:27:04  +08 [INFO] HW revision A does not support backplate LED color setting
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Drawing Image: BACKGROUND
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Drawing Text: CPU_MODEL
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Drawing Text: GPU_MODEL
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Drawing Text: RAM_MODEL
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Drawing Text: DOWNLOAD_RATE
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Drawing Text: UPLOAD_RATE
Jumaat 12 Mei 2023 10:27:04  +08 [DEBUG] Drawing Text: DISK_USAGE
Jumaat 12 Mei 2023 10:27:04  +08 [WARNING] Your GPU is not supported