mathoudebine/turing-smart-screen-python #222: AMD GPU not detected on Fedora Linux

AwekM, 2023-05-13:
**Describe the bug**  
The application doesn't detect my AMD Asrock 5700xt on Fedora Linux. Here is the output.

```
acapcap@acapcap turing-smart-screen-python-main]$ python3 main.py
13/05/2023 11:26:49 [INFO] Loading theme Cyberpunk-net from res/themes/Cyberpunk-net/theme.yaml
13/05/2023 11:26:49 [DEBUG] Auto detected COM port: /dev/ttyACM0
13/05/2023 11:26:49 [DEBUG] Using Python 3.11.3 (main, Apr  5 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)]
13/05/2023 11:26:49 [INFO] Tray icon has been displayed
13/05/2023 11:26:49 [INFO] Display reset (COM port may change)...
13/05/2023 11:26:54 [DEBUG] Auto detected COM port: /dev/ttyACM0
13/05/2023 11:26:54 [INFO] HW revision A does not support backplate LED color setting
13/05/2023 11:26:54 [DEBUG] Drawing Image: BACKGROUND`
13/05/2023 11:26:54 [DEBUG] Drawing Text: CPU_MODEL
13/05/2023 11:26:54 [DEBUG] Drawing Text: GPU_MODEL
13/05/2023 11:26:54 [DEBUG] Drawing Text: RAM_MODEL
13/05/2023 11:26:54 [DEBUG] Drawing Text: DOWNLOAD_RATE
13/05/2023 11:26:54 [DEBUG] Drawing Text: UPLOAD_RATE
13/05/2023 11:26:54 [DEBUG] Drawing Text: DISK_USAGE
13/05/2023 11:26:54 [WARNING] Your GPU is not supported yet
```

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Revision of this project [e.g. 1.1.1, `main` branch, 2.4.2 release]
 - Fedora Linux
 - Python 3.11.3
 - AMD Ryzen 5 3600
 - Asrock 5700xt

**Additional context**  
I have read [here](https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-%3A-hardware-support) that AMD GPU on Linux has limited support with this application. Is this still true or is there any other way to make it detect? 
I also have read these on the requirements.txt;

```
# Following packages are for AMD GPU on Linux
# Note: you may need to manually install Cython first
Cython~=0.29.32; sys_platform=="linux"
pyamdgpuinfo~=2.1.3; sys_platform=="linux" and python_version < "3.11" # Wheel package not yet available for Python 3.11+
```

So I tried to install cython and pyamdgpuinfo;

```
[acapcap@acapcap ~]$ python3 -m pip install cython
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: cython in ./.local/lib/python3.11/site-packages (0.29.34)
[acapcap@acapcap ~]$ python3 -m pip install pyamdgpuinfo
Defaulting to user installation because normal site-packages is not writeable
Collecting pyamdgpuinfo
  Using cached pyamdgpuinfo-2.1.3.tar.gz (21 kB)
  Preparing metadata (setup.py) ... done
Using legacy 'setup.py install' for pyamdgpuinfo, since package 'wheel' is not installed.
Installing collected packages: pyamdgpuinfo
  Running setup.py install for pyamdgpuinfo ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for pyamdgpuinfo did not run successfully.
  │ exit code: 1
  ╰─> [19 lines of output]
      running install
      /usr/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-311
      creating build/lib.linux-x86_64-cpython-311/pyamdgpuinfo
      copying pyamdgpuinfo/__init__.py -> build/lib.linux-x86_64-cpython-311/pyamdgpuinfo
      running build_ext
      building 'pyamdgpuinfo._pyamdgpuinfo' extension
      creating build/temp.linux-x86_64-cpython-311
      creating build/temp.linux-x86_64-cpython-311/pyamdgpuinfo
      gcc -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/usr/include/libdrm -I/usr/include/python3.11 -c pyamdgpuinfo/_pyamdgpuinfo.c -o build/temp.linux-x86_64-cpython-311/pyamdgpuinfo/_pyamdgpuinfo.o
      pyamdgpuinfo/_pyamdgpuinfo.c:25:10: fatal error: Python.h: No such file or directory
         25 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pyamdgpuinfo

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

mathoudebine, 2023-05-14:
Hi, there is currently an issue with the pyamdgpuinfo library.
Pre-built packages are not available for Python 3.11 because the developer has not uploaded them to pip repositories
This mean the package will have to be built on your computer by pip. To avoid build errors like you have, i disabled the installation of pyamdgpuinfo for users with Python 3.11+ in requirements.txt

You have two choices: you can either downgrade your Python version to 3.10 if possible with your distribution, or build pyamdgpuinfo manually using pip like you are doing.

For the error you have, you can refer to this stackoverflow question regarding dependencies to built Python libraries: https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory

mathoudebine, 2023-05-15:
You will also need to install `libdrm-dev` or `libdrm-devel` to be able to build pyamdgpuinfo with pip

AwekM, 2023-05-15:
Hi thank you for the solution.

Really appreciate your help.

Here's what I did.

`sudo dnf install python3-devel`

`sudo dnf install libdrm-devel`

`pip install pyamdgpuinfo`

closed 2023-05-15 by AwekM