mathoudebine/turing-smart-screen-python #230: Smart Screen has stopped working

wapiti59, 2023-05-15:
It has been working flawlessly for over a year now.  Then, all of a sudden, it has stopped.  Maybe because of a Windows update (win 11) I don't know.  I have tried reinstalling, everything I can think of.  I am running Python 3.I0. was finally able to get the config screen to come up, and have tried EVERY option in there, but it won't come up.  I have a Turing Smart Screen, I think it's 5 inch.  Can you help?

mathoudebine, 2023-05-16:
Hello, i'm gonna need more details. What happens when you run the program from a command line? Can you post the output here?
If you can check your screen model, the 5" is not currently supported by this project so either you have a 3.5" or you were not using this project but another software

wapiti59, 2023-05-16:
When I run it from the command line (DOS Prompt as admin in Win 11), I see a window flash, and then get the following output:

C:\Users\turing-smart-screen-python-main>python main.py
5/16/2023 11:05:34 AM [INFO] Loading theme LandscapeMagicBlue from res/themes/LandscapeMagicBlue/theme.yaml
5/16/2023 11:05:34 AM [DEBUG] Static COM port: COM3
System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\turing-smart-screen-python-main\main.py", line 64, in <module>
    import library.scheduler as scheduler
  File "C:\Users\turing-smart-screen-python-main\library\scheduler.py", line 28, in <module>
    import library.stats as stats
  File "C:\Users\turing-smart-screen-python-main\library\stats.py", line 45, in <module>
    import library.sensors.sensors_librehardwaremonitor as sensors
  File "C:\Users\turing-smart-screen-python-main\library\sensors\sensors_librehardwaremonitor.py", line 40, in <module>
    clr.AddReference(lhm_dll)
System.IO.FileLoadException: Could not load file or assembly 'file:///C:\Users\turing-smart-screen-python-main\external\LibreHardwareMonitor\LibreHardwareMonitorLib.dll' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515)
File name: 'file:///C:\Users\turing-smart-screen-python-main\external\LibreHardwareMonitor\LibreHardwareMonitorLib.dll' ---> System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)
   at System.Reflection.Assembly.LoadFrom(String assemblyFile)
   at Python.Runtime.AssemblyManager.LoadAssemblyFullPath(String name)
   at Python.Runtime.CLRModule.AddReference(String name)

When I run configure.py, the configuration window comes up, but when I enter any values it too errors out trying to run the main program.  As I said, this has been running flawlessly for well over a year, and only recently stopped, so this must be a 3.5" screen..

mathoudebine, 2023-05-16:
This is a known issue that should be fixed by following this Troubleshooting page: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Troubleshooting#windows-cannot-load-librehardwaremonitordll-loadfromremotesources

mathoudebine, 2023-05-16:
Just to be sure this is the correct Python path (in case you may have severl Python versions installed) can you run these commands:
```
where python
where python3
```
The output will indicate which `python.exe` is used

wapiti59, 2023-05-16:
Just saved them in there,same output/result

mathoudebine, 2023-05-16:
Just to sum-up the steps in case you missed something:
- Run `where python3` then create **python3**.exe.config file in the same folder than `python3.exe`
- Run `where python` then create **python**.exe.config file in the same folder than `python.exe`
- Run `where pythonw` then create **pythonw**.exe.config file in the same folder than `python3`

All the 3 .config files must contain the expected content:
``` xml
<configuration>  
   <runtime>  
      <loadFromRemoteSources enabled="true"/>  
   </runtime>  
</configuration>  
```

wapiti59, 2023-05-16:
Done and exactly the same output and result

mathoudebine, 2023-05-24:
If you still have the issue, you can try to go to the `external/LibreHardwareMonitor` folder of this project, then for each of the two .dll file do a right-click > Properties then click (or check) `Unblock`:  
[image]

mathoudebine, 2023-05-25:
Thanks, i added the process to Unlock files in the documentation for other users.

If you see nothing on the screen it may be a configuration issue:
First, check which smart screen you have (Turing or XuanFang) using this page: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Hardware-revisions
Then run `python3 configure.py` to start the configuration wizard, check if the selected smart screen model and the brightness level is correct

mathoudebine, 2023-06-27:
Closing this issue for inactivity, if you have some additional issues feel free to open a new one!

closed 2023-06-27 by mathoudebine

wapiti59, 2023-06-27:
Hi sir.  It kinda slipped off my plate, sorry.  Still can’t get it to work.

 

*	Doug Brown
*	303-981-4908