mathoudebine/turing-smart-screen-python #244: ![Gaming0_generated](https://user-images.githubusercontent.com/132712795/236700519-93461b5a-1200-42af-a27d-ee00c330b82c.jpg)

Citywolf01, 2023-05-26:
[image]
if you more into this then you can complete my ones

_Originally posted by @Onslau in https://github.com/mathoudebine/turing-smart-screen-python/discussions/218#discussioncomment-5831969_

Onslau, 2023-05-26:
Whats the issue its a canvas just a backround my new ones are by roby and me and need to be fixed and converted

Onslau, 2023-05-26:
https://github.com/mathoudebine/turing-smart-screen-python/discussions/217
https://github.com/mathoudebine/turing-smart-screen-python/discussions/218
we are two men he does it with phyton i use the blankets to produce

Onslau, 2023-05-26:
size must be 480x320 you need to make shapes in paint net adobe etc and us them with turning screen software to fill it out

mathoudebine, 2023-05-26:
Hello, i think you clicked on "Reference in new issue" by mistake so i'm closing this issue, because there is nothing more than the original comment.
You can use the [Discussion space](https://github.com/mathoudebine/turing-smart-screen-python/discussions) to discuss about themes and ask questions

closed 2023-05-26 by mathoudebine