mathoudebine/turing-smart-screen-python #247: Image corruption on linux

nukdokplex, 2023-05-28:
**Describe the bug**  
I getting a corrupted image on screen randomly. To fix this i only can stop service, reconnect screen hard and start service again. What can i do?

**Screenshots / photos of the Turing screen**  
[image]

**Logs**
```мая 28 20:42:08 ndp-desktop python3[616]: 28.05.2023 20:42:08 [WARNING] Tray icon is not supported on your platform
мая 28 20:42:08 ndp-desktop python3[616]: 28.05.2023 20:42:08 [INFO] Display reset (COM port may change)...
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Auto detected COM port: /dev/ttyACM0
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [INFO] HW revision A does not support backplate LED color setting
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Image: BACKGROUND
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: CPU_MODEL
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: GPU_MODEL
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: RAM_MODEL
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: DOWNLOAD_RATE
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: UPLOAD_RATE
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [DEBUG] Drawing Text: DISK_USAGE
мая 28 20:42:13 ndp-desktop python3[616]: 28.05.2023 20:42:13 [INFO] Detected AMD GPU(s)
мая 28 20:50:44 ndp-desktop python3[616]: Exception in thread Queue_Handler:
мая 28 20:50:44 ndp-desktop python3[616]: Traceback (most recent call last):
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 621, in write
мая 28 20:50:44 ndp-desktop python3[616]:     n = os.write(self.fd, d)
мая 28 20:50:44 ndp-desktop python3[616]: OSError: [Errno 5] Ошибка ввода/вывода
мая 28 20:50:44 ndp-desktop python3[616]: During handling of the above exception, another exception occurred:
мая 28 20:50:44 ndp-desktop python3[616]: Traceback (most recent call last):
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
мая 28 20:50:44 ndp-desktop python3[616]:     self.run()
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/threading.py", line 953, in run
мая 28 20:50:44 ndp-desktop python3[616]:     self._target(*self._args, **self._kwargs)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/scheduler.py", line 74, in wrap
мая 28 20:50:44 ndp-desktop python3[616]:     scheduler.run()
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/sched.py", line 151, in run
мая 28 20:50:44 ndp-desktop python3[616]:     action(*argument, **kwargs)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/scheduler.py", line 64, in periodic
мая 28 20:50:44 ndp-desktop python3[616]:     action(*actionargs)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/scheduler.py", line 163, in QueueHandler
мая 28 20:50:44 ndp-desktop python3[616]:     f(*args)
мая 28 20:50:44 ndp-desktop python3[616]:   File "/opt/turing/library/lcd/lcd_comm.py", line 99, in WriteData
мая 28 20:50:44 ndp-desktop python3[616]:     self.lcd_serial.write(bytes(byteBuffer))
мая 28 20:50:44 ndp-desktop python3[616]:   File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 655, in write
мая 28 20:50:44 ndp-desktop python3[616]:     raise SerialException('write failed: {}'.format(e))
мая 28 20:50:44 ndp-desktop python3[616]: serial.serialutil.SerialException: write failed: [Errno 5] Ошибка ввода/вывода
```

**Environment:**  
 - Project 2.4.2
 - Manjaro Linux latest
 - Python 3.10
 - Intel Core i7 6700K, Radeon RX6700XT

mathoudebine, 2023-05-29:
Hello, it seems different than the bug for MacOS users because they do not have errors on the terminal.
For you it seems there is an issue at some point when sending data to your screen.
If you need to reconnect the screen it may be a problem with the screen.
Did you test with other OS?

nukdokplex, 2023-05-31:
@mathoudebine 
Right now I can't test with other os. I have noticed that the image is corrupted under heavy load of my computer, for example when I start some game but I'm not sure about that. 

For example, I spent the whole day yesterday in VS code, Discord, Google Chrome and a couple of Docker containers were running. On average, the processor load was around 30% and the screen was not distorted even once.

Now I'm guessing that this may be due to voltage drops in the USB, but I'm not sure. Perhaps someone had this problem, it would be interesting to listen to people with a similar situation.

nukdokplex, 2023-05-31:
Update! Screen corrupted again, now without any error.

```
-- Boot 5e0a16a968fe4c4e98e156cbfd563afd --
июн 01 01:24:05 ndp-desktop systemd[1]: Started Turing Smart Screen.
июн 01 01:24:05 ndp-desktop python3[630]: 01.06.2023 01:24:05 [INFO] Loading theme Cyberpunk from res/themes/Cyberpunk/theme.yaml
июн 01 01:24:05 ndp-desktop python3[630]: 01.06.2023 01:24:05 [DEBUG] Auto detected COM port: /dev/ttyACM0
июн 01 01:24:05 ndp-desktop python3[630]: 01.06.2023 01:24:05 [DEBUG] Using Python 3.10.10 (main, Mar  5 2023, 22:26:53) [GCC 12.2.1 20230201]
июн 01 01:24:05 ndp-desktop python3[630]: 01.06.2023 01:24:05 [WARNING] Tray icon is not supported on your platform
июн 01 01:24:05 ndp-desktop python3[630]: 01.06.2023 01:24:05 [INFO] Display reset (COM port may change)...
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [DEBUG] Auto detected COM port: /dev/ttyACM0
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [INFO] HW revision A does not support backplate LED color setting
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [DEBUG] Drawing Image: BACKGROUND
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [DEBUG] Drawing Text: CPU_MODEL
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [DEBUG] Drawing Text: GPU_MODEL
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [DEBUG] Drawing Text: RAM_MODEL
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [DEBUG] Drawing Text: DISK_USED
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [DEBUG] Drawing Text: DISK_FREE
июн 01 01:24:10 ndp-desktop python3[630]: 01.06.2023 01:24:10 [INFO] Detected AMD GPU(s)
lines 1513-1554/1554 (END)
```

[image]

nukdokplex, 2023-06-12:
Image corruption has gone when I changed all update rates to minimum 2 secs.

closed 2023-06-12 by nukdokplex