mathoudebine/turing-smart-screen-python #261: Screen does not clear when shutting down Windows

misosoup7, 2023-06-09:
**Describe the bug**
If I shutdown the computer, then the screen gets stuck on the last update and doesn't turn off. I'm worried about burn in if it's left like that. Maybe we can force a blank when receiving SIGINT from shutdown?

**Screenshots / photos of the Turing screen**
[image]

**Logs**

> 6/8/2023 6:45:38 PM [INFO] Loading theme Cyberpunk-net from res/themes/Cyberpunk-net/theme.yaml
> 6/8/2023 6:45:38 PM [DEBUG] Auto detected COM port: COM3
> 6/8/2023 6:45:40 PM [DEBUG] Found LibreHardwareMonitorLib 0.9.2.0
> 6/8/2023 6:45:41 PM [INFO] Found CPU: 12th Gen Intel Core i5-12600KF
> 6/8/2023 6:45:41 PM [INFO] Found Memory: Generic Memory
> 6/8/2023 6:45:41 PM [INFO] Found Nvidia GPU: NVIDIA GeForce RTX 3070 Ti
> 6/8/2023 6:45:41 PM [INFO] Found Storage: Samsung SSD 980 PRO 1TB
> 6/8/2023 6:45:41 PM [INFO] Found Storage: Samsung SSD 840 EVO 250GB
> 6/8/2023 6:45:41 PM [INFO] Found Storage: Samsung SSD 840 EVO 250GB
> 6/8/2023 6:45:41 PM [INFO] Found Storage: Samsung SSD 980 PRO 1TB
> 6/8/2023 6:45:41 PM [INFO] Found Network interface: Bluetooth Network Connection
> 6/8/2023 6:45:41 PM [INFO] Found Network interface: Ethernet
> 6/8/2023 6:45:41 PM [INFO] Found Network interface: Wi-Fi
> 6/8/2023 6:45:41 PM [DEBUG] Using Python 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]
> 6/8/2023 6:45:44 PM [INFO] Tray icon has been displayed
> 6/8/2023 6:45:44 PM [INFO] Display reset (COM port may change)...
> 6/8/2023 6:45:49 PM [DEBUG] Auto detected COM port: COM3
> 6/8/2023 6:45:49 PM [DEBUG] Drawing Image: BACKGROUND
> 6/8/2023 6:45:49 PM [DEBUG] Drawing Text: CPU_MODEL
> 6/8/2023 6:45:49 PM [DEBUG] Drawing Text: GPU_MODEL
> 6/8/2023 6:45:49 PM [DEBUG] Drawing Text: RAM_MODEL
> 6/8/2023 6:45:49 PM [DEBUG] Drawing Text: DOWNLOAD_RATE
> 6/8/2023 6:45:49 PM [DEBUG] Drawing Text: UPLOAD_RATE
> 6/8/2023 6:45:49 PM [DEBUG] Drawing Text: DISK_USAGE
> 6/8/2023 6:45:50 PM [DEBUG] Found one supported GPU: NVIDIA GeForce RTX 3070 Ti

Nothing shows up on the log for the shutdown, you can see the next start up further down the log.

**Environment:**

Project 3.0.0
Windows 11 Pro 21H2
Python 3.11.4
Intel Core i5 12600KF
NVIDIA GeForce RTX 3070 Ti
32 GB DDR5 5200 Dual Channel

misosoup7, 2023-06-09:
I've created a workaround for Windows 11 Pro (and possibly 10 Pro) in case people are interested:

Create two scripts:
`
@echo off
start "Turing Smart Screen Python" /D <PATH TO main.py> <PATH TO pythonw.exe>\pythonw.exe main.py
`
You can get path to pythonw.exe by typing `where pythonw.exe` in Command Prompt
Save as startup.bat

Second script
`
@echo off
taskkill /im pythonw.exe
`
Save as shutdown.bat

Then in Group Policy Editor (Win+R > gpedit.msc)
Local Computer Policy > Computer Configuration > Windows Settings > Scripts (Startup/Shutdown)

In Startup > Scripts > Click "Add" and select to the startup.bat that you saved. > Ok > Ok
In Shutdown > Scripts > Click "Add" and select to the shutdown.bat that you saved. > Ok > Ok

The reason I'm not sure if will work for non-Pro windows versions is because I had to use the Group Policy Editor, which is only available on Pro. I didn't use the task scheduler is because processes started with task scheduler cannot be gracefully terminated with taskkill. I can only use it with the /f argument which leaves the screen on.

One other caveat is that the script doesn't run immediately upon start up. It looks like it runs several seconds after log on. I will do some more tests on non-pro versions of Windows but it would be great if it's fixable in the python script.

mathoudebine, 2023-06-09:
Hi, and thanks for raising this issue.
Actually the feature to turn off the screen on shutdown/logoff is already implemented in `main.py` 

This part is to create the signal handlers:
https://github.com/mathoudebine/turing-smart-screen-python/blob/1b444aaff3841a345dd8991fe109abea595f89f7/main.py#L175-L183

This part is to catch and process Windows-specific signals like `win32con.CTRL_C_EVENT, win32con.CTRL_BREAK_EVENT`:
https://github.com/mathoudebine/turing-smart-screen-python/blob/1b444aaff3841a345dd8991fe109abea595f89f7/main.py#L125-L131

What is tricky is that Windows does not send signals to running programs on shutdown/sleep/logoff actions, it instead broadcasts some system events like `WM_POWERBROADCAST`, `WM_ENDSESSION` to programs that have registered a main window and regularly process the received events.

This is what is done here by registering an invisible main window to all the system events associated with shutdown/sleep/logoff and processing them every 0.5 seconds:
https://github.com/mathoudebine/turing-smart-screen-python/blob/1b444aaff3841a345dd8991fe109abea595f89f7/main.py#L223-L255

When one of the related system events is received, it is managed here:
https://github.com/mathoudebine/turing-smart-screen-python/blob/1b444aaff3841a345dd8991fe109abea595f89f7/main.py#L134-L148

Each of this handlers call the `clean_stop()` action that:
- Adds a "turn off" command to the queue containing all commands to be sent to the display
- Stops the scheduler so that other commands cannot be added to the queue
- Waits for all remaining commands to be sent to the display (max 5s)
- Stops the Python program

So the `taskkill /im pythonw.exe` should not be necessary for the program to stop and the display to turn off on shutdown.

If there is something that doesn't work as expected, the logs could help: can you remove the Shutdown script, start the Python program if not already running, then restart your computer? In the root folder of the program there will be a `log.log` file that you can upload here, at least for the relevant lines. This would help to see what's going on.

misosoup7, 2023-06-10:
You are right that the shutdown script is not needed. However, if I run the program with task scheduler then the screen wouldn't clear correctly for some reason. And if run it with group policy then it's fine. Very weird. No difference in the log.log between the two.

But while I was testing, not the script refuse to open the port to the display now. Windows keeps detecting it as being plugged in and then disconnecting it and instantly reconnecting it whenever I try to run main.py. I can get it to stop only when I turn off the computer.

```
6/9/2023 9:27:00 PM [INFO] Loading theme Cyberpunk-net from res/themes/Cyberpunk-net/theme.yaml
6/9/2023 9:27:00 PM [DEBUG] Auto detected COM port: COM3
6/9/2023 9:27:01 PM [DEBUG] Found LibreHardwareMonitorLib 0.9.2.0
6/9/2023 9:27:05 PM [INFO] Found CPU: 12th Gen Intel Core i5-12600KF
6/9/2023 9:27:05 PM [INFO] Found Memory: Generic Memory
6/9/2023 9:27:05 PM [INFO] Found Nvidia GPU: NVIDIA GeForce RTX 3070 Ti
6/9/2023 9:27:05 PM [INFO] Found Storage: Samsung SSD 980 PRO 1TB
6/9/2023 9:27:05 PM [INFO] Found Storage: Samsung SSD 840 EVO 250GB
6/9/2023 9:27:05 PM [INFO] Found Storage: Samsung SSD 840 EVO 250GB
6/9/2023 9:27:05 PM [INFO] Found Storage: Samsung SSD 980 PRO 1TB
6/9/2023 9:27:05 PM [INFO] Found Network interface: Bluetooth Network Connection
6/9/2023 9:27:05 PM [INFO] Found Network interface: Ethernet
6/9/2023 9:27:05 PM [INFO] Found Network interface: Local Area Connection* 10
6/9/2023 9:27:05 PM [INFO] Found Network interface: Local Area Connection* 9
6/9/2023 9:27:05 PM [INFO] Found Network interface: Wi-Fi
6/9/2023 9:27:05 PM [DEBUG] Using Python 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]
6/9/2023 9:27:05 PM [INFO] Tray icon has been displayed
6/9/2023 9:27:05 PM [INFO] Display reset (COM port may change)...
6/9/2023 9:27:10 PM [DEBUG] Auto detected COM port: COM3
6/9/2023 9:27:19 PM [DEBUG] Caught Windows control event 0, exiting
6/9/2023 9:27:19 PM [INFO] Waiting for all pending request to be sent to display (5s max)...
6/9/2023 9:27:24 PM [DEBUG] (5.1s)
```
and in the command prompt it says right after `6/9/2023 9:27:10 PM [DEBUG] Auto detected COM port: COM3`:

```
Traceback (most recent call last):
  File "D:\TuringSmartScreen\turing-smart-screen-python-3.0.0\main.py", line 186, in <module>
    display.initialize_display()
  File "D:\TuringSmartScreen\turing-smart-screen-python-3.0.0\library\display.py", line 88, in initialize_display
    self.lcd.Reset()
  File "D:\TuringSmartScreen\turing-smart-screen-python-3.0.0\library\lcd\lcd_comm_rev_a.py", line 87, in Reset
    self.openSerial()
  File "D:\TuringSmartScreen\turing-smart-screen-python-3.0.0\library\lcd\lcd_comm.py", line 85, in openSerial
    self.lcd_serial = serial.Serial(lcd_com_port, 115200, timeout=1, rtscts=1)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\MyUser\AppData\Local\Programs\Python\Python311\Lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\MyUser\AppData\Local\Programs\Python\Python311\Lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "C:\Users\MyUser\AppData\Local\Programs\Python\Python311\Lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)
```

I thought the screen might have broken, but Turing's software still works fine with it after a reboot...

arthurferrai, 2023-06-12:
Just to be sure: which Turing display are you using? 3" or 5"?

misosoup7, 2023-06-12:
The 3.5".

Drahgoone, 2024-06-02:
I am having this same issue. If I close the Turing program the screen clears fine, but when I shutdown it does not. Have yet to figure out a workaround for Windows 11 Home.

FedorPRO, 2024-07-06:
Same problem, Windows 11 Pro / Release 3.5.0 / Python 3.12.4

It’s very strange, but somewhere in November-December 2023 it was still working. For about half a year now I have had to either manually disable the application or unplug the USB cable. Now I repeated the entire installation again - nothing has changed - the screen continues to light up when you put the PC to sleep. It feels like the PC falls asleep faster than it can process the code.

FedorPRO, 2024-07-06:
Surprisingly, after trying 15-20 times the screen began to go dark.

Log when the screen remained on:
```
06.07.2024 17:18:03 [DEBUG] Caught Windows window message event 536
06.07.2024 17:18:03 [INFO] Computer is going to sleep, display will turn off
06.07.2024 17:18:03 [DEBUG] Caught Windows window message event 536
06.07.2024 17:18:03 [DEBUG] Caught Windows window message event 536
06.07.2024 17:18:03 [INFO] Computer is resuming from sleep, display will turn on
```

Log when the screen goes dark
```
06.07.2024 17:30:31 [DEBUG] Caught Windows window message event 536
06.07.2024 17:30:31 [INFO] Computer is going to sleep, display will turn off
06.07.2024 17:31:00 [DEBUG] Caught Windows window message event 536
06.07.2024 17:31:00 [DEBUG] Caught Windows window message event 536
06.07.2024 17:31:00 [INFO] Computer is resuming from sleep, display will turn on
```

as you can see from the log, when the screen remained lit for some reason this block of code was processed
```
                elif wParam == win32con.PBT_APMRESUMEAUTOMATIC:
                    logger.info("Computer is resuming from sleep, display will turn on")
                    display.turn_on()
                    # Some models have troubles displaying back the previous bitmap after being turned off/on
                    display.display_static_images()
                    display.display_static_text()
```