mathoudebine/turing-smart-screen-python #263: Landscape themes not rotating on type A "Turing" smart screen on Ubuntu 22.04.2

masonlouie, 2023-06-14:
**Describe the bug**  
Using configure.py to select a Landscape theme (like LandscapeMagicBlue) renders the theme incorrectly: in portrait instead of landscape mode & with an "lines" (interlacing?) effect on the partially rendered screen (see picture). All the portrait themes work fine. None of the landscape themes work. Selecting "reverse" (instead of "classic") doesn't do anything.

**To Reproduce**  
Steps to reproduce the behavior:
1. Download/install this github repo & its requirements.
2. From the main repo directory, run: python configure.py
3. In the TK GUI, select a landscape theme like LandscapeMagicBlue.
4. Click "Save and Run". Watch the debug messages scroll by until the screen displays the incorrect rendering

**Expected behavior**  
I expect landscape themes to display in a landscape orientation.

**Screenshots / photos of the Turing screen**  
LandscapeMagicBlue rendered in portrait mode instead of landscape over bash-dark-green-gpu
[image]
This photo shows LandscapeMagicBlue (LMB) rendered on top of bash-dark-green-gpu (which does render and work correctly). The sensor values for LMB change (and are pressumably correct), but no changes to the orientation parameter will change the render orientation from portrait (no reverse or landscape). I think the byte lines are wrapping causing the interlacing/scrunched look. I'm not sure why the previous theme's background also gets rendered.

**Environment:**  
 - turing-smart-screen-python 3.0.0 (normal clone from github)
 - Ubuntu 22.04.2 LTS, Linux 5.19.0-43-generic 44~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC x86_64 x86_64 x86_64
 - Python 3.10.9
 - 3.5" "Turing Smart Screen" (type A) plugged into a USB 2.0 header on my motherboard. 
 - AMD 5700x on a EVGA x570 FTW mobo

**Additional context**  
 - I've downloaded the OEM monitor software and run it in Windows. Its themes, both portrait and landscape, work fine. So the hardware is working fine. 
 - I've tried using the Type B configuration but nothing renders (pressumably bc my smart screen is not a Type B).

MAIKS1900, 2023-06-17:
Same issue with type A on Win11 with python 3.10/3.11. 
Official program renders landscape themes perfectly fine. 
Turing-smart-screen-python 2.4.3/3.0.0/latest tries to render themes in portrait mode.

mathoudebine, 2023-06-18:
Hello, can you post the content of the `log.log` file in the root folder? If there are any errors they should appear in this file

mathoudebine, 2023-06-18:
Another test you can do is to comment this line in `display.py`: https://github.com/mathoudebine/turing-smart-screen-python/blob/1b444aaff3841a345dd8991fe109abea595f89f7/library/display.py#L88 the Reset() sometimes causes issues with some Turing displays

MAIKS1900, 2023-06-18:
Hi! 
Unplugged screen and tried Landscape6Grid. Now it has white screen on background. Here's log, seems no errors:
```
18.06.2023 20:31:16 [INFO] Loading theme Landscape6Grid from res/themes/Landscape6Grid/theme.yaml
18.06.2023 20:31:16 [DEBUG] Static COM port: COM6
18.06.2023 20:31:16 [DEBUG] Found LibreHardwareMonitorLib 0.9.2.0
18.06.2023 20:31:17 [INFO] Found CPU: AMD Ryzen 5 5600X
18.06.2023 20:31:17 [INFO] Found Memory: Generic Memory
18.06.2023 20:31:17 [INFO] Found Nvidia GPU: NVIDIA GeForce RTX 3070 Ti
18.06.2023 20:31:17 [INFO] Found Storage: WD Blue SN570 1TB
18.06.2023 20:31:17 [INFO] Found Network interface: Ethernet
18.06.2023 20:31:17 [INFO] Found Network interface: ������� ����������� Bluetooth
18.06.2023 20:31:17 [DEBUG] Using Python 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)]
18.06.2023 20:31:17 [INFO] Tray icon has been displayed
18.06.2023 20:31:17 [INFO] Display reset (COM port may change)...
18.06.2023 20:31:22 [DEBUG] Static COM port: COM6
18.06.2023 20:31:22 [DEBUG] Drawing Image: BACKGROUND
18.06.2023 20:31:22 [DEBUG] Found one supported GPU: NVIDIA GeForce RTX 3070 Ti
18.06.2023 20:31:28 [DEBUG] Caught Windows control event 0, exiting
18.06.2023 20:31:28 [INFO] Waiting for all pending request to be sent to display (5s max)...
18.06.2023 20:31:28 [DEBUG] (0.0s)
```

Also tried to comment self.lcd.Reset(), but log output is the same

MAIKS1900, 2023-06-18:
Also noticed. Reverse portrait mode doesn't work after unplugging.
I managed to run it this way:
1. Started official UsbMonitor.exe
2. Choose reverse portrait theme
3. Killed process in task manager (not built-in exit)
4. Run Turing-smart-screen-python with portrait theme

And it worked. 
So may be we have some new revisions of display and [SendCommand](https://github.com/mathoudebine/turing-smart-screen-python/blob/1b444aaff3841a345dd8991fe109abea595f89f7/library/lcd/lcd_comm_rev_a.py#L59) doesn't work correctly now(

mathoudebine, 2023-06-18:
You may have a new revision of the Turing 3.5" screen that i'm not aware of, the issue could also be on the `SetOrientation` command:
https://github.com/mathoudebine/turing-smart-screen-python/blob/45833060d4bfcb1a57f278dd25c662e790f5bf12/library/lcd/lcd_comm_rev_a.py#L110-L130
As you can see, this is the only command that has a specific format and do not use the `SendCommand()` function. Maybe something is wrong in this command, or there is a new revision of the screen that has a different command for rotation.

Can you post the exact link to the product you bought?

masonlouie, 2023-06-19:
> Hello, can you post the content of the `log.log` file in the root folder? If there are any errors they should appear in this file
> 
> I see on your photo that you have remains of the `bash-dark-green` portrait theme displayed, which should not be the case because the screen should be reset each time `main.py` is restarted. How do you stop/start the program? Can you do a test where you unplug the screen, plug it back in, then start the program with a landscape theme?

Unplugging/plugging back the screen shows the same as the above image except the bash-dark-green-gpu is replaced by a white background. I start the program with "python configuration.py" and sometimes "python main.py". I press Ctrl+C to terminate the program and then wait for the cleanup messages to finish. Sometimes I notice I have several main.py running. When I notice this, I run "killall -9 python" which stops all instances of main.py. The image I took above had only one "python main.py" at the time.

Log.log output looks like:

>    06/16/2023 02:13:15 PM [INFO] Loading theme LandscapeMagicBlue from
   res/themes/LandscapeMagicBlue/theme.yaml
   06/16/2023 02:13:15 PM [DEBUG] Auto detected COM port: /dev/ttyACM0
   06/16/2023 02:13:15 PM [DEBUG] Using Python 3.10.6 (main, May 29
   2023, 11:10:38) [GCC 11.3.0]
   06/16/2023 02:13:15 PM [WARNING] Tray icon is not supported on your
   platform
   06/16/2023 02:13:15 PM [INFO] Display reset (COM port may change)...
   06/16/2023 02:13:20 PM [DEBUG] Auto detected COM port: /dev/ttyACM0
   06/16/2023 02:13:20 PM [DEBUG] Drawing Image: BACKGROUND
   06/16/2023 02:13:20 PM [DEBUG] Drawing Text: DISK_LABEL
   06/16/2023 02:13:20 PM [DEBUG] Drawing Text: DISK_USED_LABEL
   06/16/2023 02:13:20 PM [DEBUG] Drawing Text: GPU_TEMP_UNIT
   06/16/2023 02:13:20 PM [INFO] Detected Nvidia GPU(s)
   06/16/2023 02:13:50 PM [INFO] Caught signal 2, exiting
   06/16/2023 02:13:50 PM [INFO] Waiting for all pending request to be
   sent to display (5s max)...
   06/16/2023 02:13:50 PM [DEBUG] (0.0s)

masonlouie, 2023-06-19:
Pressed closed on accident. Reopening.

masonlouie, 2023-06-19:
I bought this item at the end of May: https://www.ebay.com/itm/265809542816

Commenting out Reset() on line 88 from initialize() causes the screen to not display the LandscapeMagicBlue background image at all. I don't see any of the system stats rendered. Only a few lines are out of place on the previous background image.

I tried setting reverse landscape in UsbMonitor.exe on Win 10 (in a VM) then switching to Linux and turing-smart-screen-python. It always renders in portrait without reverse no matter what state I leave UsbMonitor.exe.

I saw SetOrientation() and was grateful someone took the time to figure that out. Thank you.

I ran lsusb -v -s 1:3 (it's dev 3 on bus 1) and its output looks like:

>    Bus 001 Device 003: ID 1a86:5722 QinHeng Electronics UsbMonitor
   Device Descriptor:
      bLength                18
      bDescriptorType         1
      bcdUSB               2.00
      bDeviceClass            2 Communications
      bDeviceSubClass         0
      bDeviceProtocol         0
      bMaxPacketSize0        64
      idVendor           0x1a86 QinHeng Electronics
      idProduct          0x5722
      bcdDevice            1.00
      iManufacturer           1 2017-2-25
      iProduct                2 UsbMonitor
      iSerial                 3 USB35INCHIPSV2
      bNumConfigurations      1
      Configuration Descriptor:
        bLength                 9
        bDescriptorType         2
        wTotalLength       0x0043
        bNumInterfaces          2
        bConfigurationValue     1
        iConfiguration          0
        bmAttributes         0xa0
          (Bus Powered)
          Remote Wakeup
        MaxPower              500mA
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        0
          bAlternateSetting       0
          bNumEndpoints           1
          bInterfaceClass         2 Communications
          bInterfaceSubClass      2 Abstract (modem)
          bInterfaceProtocol      1 AT-commands (v.25ter)
          iInterface              0
          CDC Header:
            bcdCDC               1.10
          CDC Call Management:
            bmCapabilities       0x00
            bDataInterface          0
          CDC ACM:
            bmCapabilities       0x02
              line coding and serial state
          CDC Union:
            bMasterInterface        0
            bSlaveInterface         1
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x81  EP 1 IN
            bmAttributes            3
              Transfer Type            Interrupt
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0008  1x 8 bytes
            bInterval             255
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        1
          bAlternateSetting       0
          bNumEndpoints           2
          bInterfaceClass        10 CDC Data
          bInterfaceSubClass      0
          bInterfaceProtocol      0
          iInterface              0
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x82  EP 2 IN
            bmAttributes            2
              Transfer Type            Bulk
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               0
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x02  EP 2 OUT
            bmAttributes            2
              Transfer Type            Bulk
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               0
   Device Status:     0x0000
      (Bus Powered)

MAIKS1900, 2023-06-19:
As for me, bought at May on https://aliexpress.ru/item/1005003503595215.html

Also ran lsusb (on WSL) and its output almost the same as [masonlouie](https://github.com/masonlouie) has:
```
Bus 001 Device 002: ID 1a86:5722 QinHeng Electronics UsbMonitor
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x5722
  bcdDevice            1.00
  iManufacturer           1
  iProduct                2
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          0
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

masonlouie, 2023-06-19:
Would a USB packet dump help?

Attached is a zip of a Wireshark pcap, a USB packet sniffing session, of UsbMonitor.exe. 
[UsbMonitor_dump.zip](https://github.com/mathoudebine/turing-smart-screen-python/files/11786060/UsbMonitor_dump.zip)

I pressed the "Run" button with a landscape theme selected. I checked that the screen was working correctly and then let it run for a few seconds (stats updating) before stopping the sniffer and UsbMonitor. There might be some mouse packets from the virtual mouse/keyboard.

Setup: USBPcap (recommended by Wireshark) on Win10 VirtualBox VM on Ubuntu 22.04

masonlouie, 2023-06-19:
Changing SetOrientation's byteArray's size to 16 causes LMB to display in landscape. Wonderful! 

However "reverse" still doesn't work.

masonlouie, 2023-06-19:
Upon further inspection with @MAIKS1900's byteArray change, LMB reverse is rendering correctly. Great! I'm so used to seeing it wrong that when it's correct, I think it's wrong.

Also, when I run "python configuration.py" and press "Save and Run", configuration.py finishes but spawns a "python main.py" in the background which is why I sometimes get many instances of main.py running (and causing me confusion). A message from configuration.py like "Spawning main.py in background" would be nice.

mathoudebine, 2023-06-20:
Thanks for investigating, I just tested and changing the size to 16 has no impact on older models, rotation still works fine.
For the new packet with the command '7a' before the rotation command, I have no idea what it is supposed to do. Since it works for you 2 without it, I think we can ignore it for now.

closed 2023-06-20 by masonlouie

mathoudebine, 2023-06-23:
Hey @masonlouie @MAIKS1900 I'm investigating about these new commands on Turing smart screens. I just found that some manufacturers made smart screens based on the original Turing serial protocol but with additional commands.

I wanted to ask you: what is the OEM Windows software for the screen you bought?
Is it `Turing Smart Screen` like this preview ? 
[image]

or UsbPCMonitor like this one ?
[image]

Thanks!

masonlouie, 2023-06-23:
It looks like the first image, the Turing Smart Screen.

MAIKS1900, 2023-06-23:
Same, like first one