mathoudebine/turing-smart-screen-python #264: 8.8 inch screen support

arthurv, 2023-06-17:
I've got the 3.5 inch screen running nicely, and a 5 inch screen is on the way.

However, it seems an 8.8 inch screen is also available, any plans for supporting this?
https://www.aliexpress.com/item/1005005538293555.html

It looks like the perfect screen (in portrait mode) to display a to-do list, for example.

mathoudebine, 2023-06-18:
Hi! So there is now an 8.8 inch screen, great news but I'll be ruined if they release new models every 6 months 😄
If the protocol is the same as the 5.5 inch screen (which seems to be the case becasue the configuration software looks the same) then supporting it will be quick and easy.
But if they develop a new protocol for the 8.8 inch display, it will require to analyze serial frames like people did for the 5.5 inch so it will take longer.
Anyway, I will order one and we'll see! I would be happy if we can add it to the list of supported displays

rastersoft, 2023-09-22:
Hi. I see that there is a branch for 8.8 inch screens. I have one. Can I help?

Paradoxum, 2023-12-01:
Glad people are finding this and getting involed. Is there a discord? would be good to have a channel for example talking about themes etc.

ferni7, 2024-01-01:
Hey, does the 8 inch branch work? If so I'll order one and have a play...

roid1995, 2024-01-23:
Hi, i have buy this screen it'is awesome. Can you support him?

WanaGo, 2024-04-19:
Hi, has there been any progress towards support for the 8.8 inch display? Thanks

eamanola, 2024-05-24:
+1 interest

SRadyukov, 2024-06-02:
interest for 2.1 inch support
i tested branch https://github.com/mathoudebine/turing-smart-screen-python/tree/feature/264-88-inch-screen-support
device detected but image currupted - look like smt wrong with bitmap send 
i dont have protocol spec for this device, how i can help?

SolarBlackhole, 2024-06-17:
Looking to replace my aida64 8.8 screen with a turing one, definitely interesting in seeing this screen size supported.

Tharkus, 2024-06-18:
+1 interest

WanaGo, 2024-07-11:
Hi again @mathoudebine - Do you have any plans to add 8.8 support any time soon, or is this still some time off? I would love to use this, the standard software is a challenge

scottrlarson, 2024-07-28:
Looking forward to this too. I might have to look into figuring this out I want to turn my 8.8 in screen into a command line program output display for my data recovery jobs.

Jim-Wright, 2024-08-11:
Just ordered an 8.8 and am interested also.  Haven’t seen anyone mention they’ve tested this branch, guess I’ll have a look when mine gets here.

Sonny-Crockett, 2024-08-16:
+1 to 8"

credmond81, 2024-09-01:
Also hoping for 8.8" support soon. Really cool stuff!

Jdmoto, 2024-09-27:
Any luck on supporting the 8.8 inch?

hossamsy, 2024-10-06:
If that 8.8 inch usb-only screen work... it is great looking and seems faster than the Turzx 3.5 laggy one

nreinartz, 2024-10-06:
Also got mine some weeks ago and started working on supporting it. Protocol was slightly changed but I am mostly finished and am currently testing all the functionality by making some example themes that already come with the turzx software. I would also be interested in adding the support for dynamic background videos but it doesn't seem like that has been done for any type of screen yet, so this might be a little more work.

Will do a pr as soon as I am done with the basic support.

nreinartz, 2024-10-07:
I don't even know what this is, sorry.

Basic Support = Hardware stats
Dynamic Background Videos = Hardware stats with a video as background instead of a static image

hossamsy, 2024-10-07:
Simhub is simracing app like Aida but it shows the racing game stats on a usb screen, they have support for 3.5 and 5 inches turing in it, but not the 8 inch, it is what all sim racers use to connect dashboard screens and read games data, that is where the money is today

Grotok89, 2024-12-04:
> Also got mine some weeks ago and started working on supporting it. Protocol was slightly changed but I am mostly finished and am currently testing all the functionality by making some example themes [...]

Unfortunately i could not get it to work
and there is also no test theme for 8.8 right?
tried to modify a theme.yaml and background.
but then i get

Traceback (most recent call last):
  File "C:\lcd\main.py", line 189, in <module>
    display.initialize_display()
  File "C:\lcd\library\display.py", line 95, in initialize_display
    self.lcd.InitializeComm()
  File "C:\lcd\library\lcd\lcd_comm_rev_c.py", line 229, in InitializeComm
    self._hello()
  File "C:\lcd\library\lcd\lcd_comm_rev_c.py", line 211, in _hello
    if response.startswith(SubRevision.FIVEINCH.value):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: startswith first arg must be str or a tuple of str, not bytearray

maylton, 2024-12-14:
I want to help with supporting the turing 2.1 screen. How can I help?

Mumrikken, 2024-12-18:
+1 on support for the 2.1 inch screen!

Baho73, 2025-01-17:
+1 8.8 inch screen

maylton, 2025-01-17:
Any update on the 2.1 support?

kaishi00, 2025-01-18:
+1 for 8.8 support! Recently migrated to linux from windows and kinda need this to work :)

mathoudebine, 2025-02-20:
Hi all
Just to let you know that the support for Turing 2.1" and 8.8" displays is planned for the next release!
The pull request is here #690 
For those who can't wait, you can already try the branch `feature/264-88-inch-screen-support`

Tharkus, 2025-02-20:
> Olá a todos Só para você saber que o suporte para telas Turing 2.1" e 8,8 está planejado para o próximo lançamento! O pull request está aqui [? 690](https://github.com/mathoudebine/turing-smart-screen-python/pull/690) Para aqueles que não podem esperar, você já pode tentar o branch `feature/264-88-inch-screen-support`

Thank you very much, I've been waiting for this news for months.

ferni7, 2025-02-21:
Nice! Anyone found it cheaper than the link in the OP abovew?

closed 2025-02-21 by mathoudebine

mathoudebine, 2025-02-21:
New release 3.9.0 is out with support for 8.8" and 2.1" screens: https://github.com/mathoudebine/turing-smart-screen-python/releases/tag/3.9.0