mathoudebine/turing-smart-screen-python #265: No Supported GPU detected on Nvidia GPU

BlankTaken, 2023-06-18:
**Describe the bug**  
No GPU is detected when starting the program. Rest of statistics show up except for the GPU

**To Reproduce**  
Steps to reproduce the behavior:

Launch the program after saving and running, screen turns on but is missing GPU stats and console says no supported gpu detected.

**Expected behavior**  

Gpu is detected and stats for GPU appear.

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the Turing screen to help explain your problem.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Revision of this project: 3.0
 - OS: Windows 10
 - Hardware: GPU: Gigabyte 3080 Gaming OC

**Additional context**  
Add any other context about the problem here.

BlankTaken, 2023-06-18:
Terminal:

6/18/2023 2:25:38 AM [DEBUG] Auto detected COM port: COM7
6/18/2023 2:25:39 AM [DEBUG] Found LibreHardwareMonitorLib 0.9.2.0
6/18/2023 2:25:41 AM [INFO] Found CPU: AMD Ryzen 7 5800X3D
6/18/2023 2:25:41 AM [INFO] Found Memory: Generic Memory
6/18/2023 2:25:41 AM [INFO] Found Storage: WDC WD10EZEX-08WN4A0
6/18/2023 2:25:41 AM [INFO] Found Storage: SanDisk SDSSDH3512G
6/18/2023 2:25:41 AM [INFO] Found Storage: ST1000DM003-1CH162
6/18/2023 2:25:41 AM [INFO] Found Storage: PCIe SSD
6/18/2023 2:25:41 AM [INFO] Found Storage: CT1000BX500SSD1
6/18/2023 2:25:41 AM [INFO] Found Network interface: Bluetooth Network Connection 3
6/18/2023 2:25:41 AM [INFO] Found Network interface: Ethernet
6/18/2023 2:25:41 AM [INFO] Found Network interface: Local Area Connection* 11
6/18/2023 2:25:41 AM [INFO] Found Network interface: Local Area Connection* 8
6/18/2023 2:25:41 AM [INFO] Found Network interface: Wi-Fi 3
6/18/2023 2:25:41 AM [DEBUG] Using Python 3.8.5 (tags/v3.8.5:580fbb0, Jul 20 2020, 15:43:08) [MSC v.1926 32 bit (Intel)]
6/18/2023 2:25:41 AM [INFO] Tray icon has been displayed
6/18/2023 2:25:41 AM [INFO] Display reset (COM port may change)...
6/18/2023 2:25:46 AM [DEBUG] Auto detected COM port: COM7
6/18/2023 2:25:46 AM [DEBUG] Drawing Image: BACKGROUND
6/18/2023 2:25:46 AM [DEBUG] Drawing Text: DISK_FREE_LABEL
6/18/2023 2:25:46 AM [DEBUG] Drawing Text: DISK_TOTAL_LABEL
6/18/2023 2:25:47 AM [WARNING] No supported GPU found

closed 2023-06-18 by BlankTaken

mathoudebine, 2023-06-19:
Hello, I see you closed your issue, is it resolved? If so, how did you manage to solve it?

BlankTaken, 2023-06-19:
> Hello, I see you closed your issue, is it resolved? If so, how did you manage to solve it?

Hello, apologies for that, I switched over to a version of python (3.11) that was provided by the Microsoft store (if that makes any difference) and it resolved the issue. 

I was running Python 3.8.5 previously and it gave me the error stated.

Issue has been resolved and working flawlessly.