mathoudebine/turing-smart-screen-python #269: IO errno 5.

nukdokplex, 2023-06-21:
**Describe the bug**  
IO errno 5.

**To Reproduce**  
I don't know how to reproduce this. It's random. For example last error was in a moment when i quit a game.

**Expected behavior**  
No IO error OR not static screen. Maybe this IO error we can just ignore and don't stop queue?

**Screenshots / photos of the Turing screen**  
So this is error in logs:
[image]
And here is ``ls /dev/tty*``
[image]

**Environment:**  
 - 1b444aaff3841a345dd8991fe109abea595f89f7
 - Manjaro latest
 - Python 3.11
 - MSI Z170A GAMING M5, Intel Core i7 6700K, AMD Radeon RX6700XT 

**Additional context**  
Increased update time to minimum 2 secs in theme. ``systemctl restart turing`` restarts screen successfully and it continues working.

nneul, 2023-06-21:
Seeing this as well. I'd be fine if it would _actually crash_ - but the way it's currently behaving, it just hangs and doesn't terminate with the exception. If it exited the service could at least restart.

nukdokplex, 2023-06-21:
@nneul but exception doesn't repeating. I think thread is actually stopped and doesn't throw this exception again. So there is two possible explanations:

1. Thread is actually stopped.
2. Thread writes serial without any errors, but screen doesn't update.

I could test this if I could somehow turn off screen restart during script initialization. If the first statement is true, then the screen should continue to update, if not, then the second statement is true. Please check if my logic is correct?

nneul, 2023-06-21:
Probably easy enough to add a bit of debugging to write out a plain text debug message every N passes through the thread that is doing the serial writes. I think your logic makes sense.

nneul, 2023-06-21:
It actually looks like it's being fired off as a repeating invoked function, and not a looped thread. I've added some logging to my instance, and will see if there is anything useful in output.

nukdokplex, 2023-06-22:
Ok, so, i think that i can reproduce this strange bug. I quit a videogame (Prey 2017 via lutris proton if necessary) and screen was stuck instantly. I really don't know htf this s is connected, but... it is!  And i don't know is it specific problem of proton, of prey or lutris. If anyone have some guesses i will listen very carefuly. So... WTF???

nneul, 2023-06-22:
Not familiar with the game - but - does it use a headset? Maybe there is some sort of USB bus reset or other operation that is making the turing 'stumble'? 

No video games in my case. 

Does it happen every time you exit the game?

nukdokplex, 2023-06-22:
@nneul yes, every time. And i don't use usb headset - it's jack3.5.

nneul, 2023-06-22:
[image]

Meaning of that output - I wrapped each invocation of the callback function in the queue with prints of < and > respectively. The thread is either terminating or hanging in the middle of invocation of the callback function.

nneul, 2023-06-22:
Could potentially be handled by adding a try around the queue callback function, and then if it traps an IO error, set a global to get it to exit the entire app or reestablish the serial connection.

nukdokplex, 2023-06-24:
@nneul i'm waiting for that commit

nneul, 2023-06-24:
Not sure what you mean - I don't have any fix - I'm just another user of this -- not the developer. Above was just a possible suggestion on maybe how to address it - not something I'm directly working on.

mathoudebine, 2023-06-26:
@nukdokplex and @nneul I tried to make a fix for the issue you are encountering: on I/O error the program will close and reopen the COM port before a retry.
Can you test the branch `fix/269-io-errno-5` and tell me if the fix works for you? I cannot reproduce the issue on my side, so I can't tell if it would work or not. Thanks!

nneul, 2023-06-27:
I've installed it, will see if it looks, but it's relatively infrequent on my system. If @nukdokplex can easily reproduce that's going to be best test case.

nukdokplex, 2023-06-27:
@nneul @mathoudebine Unfortunately, this fix didn't work. The exception is still thrown, but the script does not exit.

[image]

nukdokplex, 2023-06-27:
@nneul @mathoudebine upd: i'm sorry, didn't checkout right commit. Seems like it's working. 

[image]

After SerialException it reopens and screen continues updating.

AwekM, 2023-06-27:
Can confirm the test branch works for me as well.

closed 2023-06-27 by mathoudebine

mathoudebine, 2023-06-28:
Maybe the COM port auto discovery isn't necessary in this case: the program should try to open the same serial port, if it fails then the program will stop because it means COM port has changed/screen is disconnected so it is a valid stop case.
I'll make the changes, maybe in your case it will work without program exiting

nneul, 2023-06-29:
```06/28/23 16:10:49 [ERROR] Cannot open COM port /dev/ttyACM0: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
```

I suspect in my case what may be going on is that the device is blipping and then coming back. Something does appear to be causing a USB reset on my system causing it to disconnect and reconnect, but not lose power.