mathoudebine/turing-smart-screen-python #282: How to pick Drive?

geekm0nkey, 2023-07-10:
Using the cyberpunk theme that shows drive usage/free and percent. I couldn't find anywhere in the UI or Yaml to pick which drive I want the stats to show for. How can you pick? It seems to be choosing my second drive. Drive D:

Thank you.

mathoudebine, 2023-08-16:
Hello! It is not possible to select the drive to use for now using YAML, this will come in a future update! In the meantime you can edit the Python code like that:
https://github.com/mathoudebine/turing-smart-screen-python/blob/965651189eb9f313572981fb11a57a538eae9d01/library/sensors/sensors_librehardwaremonitor.py#L345
change this line to:
```python
        return psutil.disk_usage("C:\\").percent
```
you will have to apply this change to the 2 other methods in the Disk class

waynehuang85, 2024-08-29:
adding a +1 for this feature, I am running Turing on my Synology (great to show CPU, Network Up/Down in realtime!) but it shows only 1st drive not the actual storage pool I am hoping it to show

```
 df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/md0        2.3G  1.2G  1.1G  54% /
...
/dev/md2        8.8T  2.5T  6.3T  29% /volume1
...
```

will try the workaround above ! thanks!

waynehuang85, 2024-08-30:
still not showing the desired disk stat, any thought? (showing 1G and around 54% in disk_label regardless in the LandscapeMagicBlue theme)

1. using python importing psutil and quoting the  mouted drive matching from df -h showing the correct disk stat

```
>>> psutil.disk_usage('/volume1').percent
28.1
>>> psutil.disk_usage('/volume1').free
6901607559168
>>> psutil.disk_usage('/volume1').used
2694450302976

```

2.  In the sensors_librehardwaremonitor.py file, modified as suggested in this issue

```
class Disk(sensors.Disk):
    @staticmethod
    def disk_usage_percent() -> float:
        return psutil.disk_usage('/volume1').percent

    @staticmethod
    def disk_used() -> int:  # In bytes
        return psutil.disk_usage('/volume1').used

    @staticmethod
    def disk_free() -> int:  # In bytes
        return psutil.disk_usage('/volume1').free
```

waynehuang85, 2024-08-30:
fixed! 
the change should be made in the sensors_python.py file instead and it works in my case ! thank you

kkm, 2024-11-05:
@mathoudebine 

I tried with Windows 11 and I changed

` return psutil.disk_usage("D:\\").percent`

and theme shows no change and it remains root hard drive.

https://github.com/mathoudebine/turing-smart-screen-python/blob/main/library/sensors/sensors_python.py#L448

kkm, 2024-11-05:
I found here:
https://github.com/mathoudebine/turing-smart-screen-python/blob/main/library/sensors/sensors_librehardwaremonitor.py#L449

I am not sure..

mathoudebine, 2024-11-13:
@kkm If you use Windows, the change needs to be done in `library/sensors/sensors_librehardwaremonitor.py`
If you use Linux/MacOS, the change needs to be done in `library/sensors/sensors_python.py`
In doubt you can apply the change on both files

GoDllkE, 2025-05-21:
it's only possible to change it if using source code? I'm using the "compiled version for windows" AKA .exe and found no file to change that.